prusa3d/PrusaSlicer #13797: 2.9.0 rc1 Sending 3D-file from Mac with Fusion 360 to Prusa Slicer > Prusa Connect > MK4S - File name error in Prusa Slicer only..

BernhardFe, 2024-12-19:
### Description of the bug

When I send a model from Mac with Fusion360 to Prusa Slicer, the name will displayed in a cryptic number: "ff59fd0b-05e8-4249-8a4b-c43187ba70cd" instead of the name of the device. But only in Slicer. In Prusa connect and on printer the name is correct shown after slicing and transfer. This happens not in the beta version before.

### Project file & How to reproduce

See above.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.9.0 rc1

### Operating system

Mac OS latest, Fusion 360 latest

### Printer model

MK4 S

BernhardFe, 2024-12-19:
Sending from Fusion 360 3MF format, but it happens in all formats.

TerPuc, 2024-12-19:
Dear @BernhardFe, thank you for your message. We were not able to reproduce such state. Can I kindly ask you to provide us with a screenshot or a video of your issue? Thank you!

BernhardFe, 2024-12-19:
Dear @TerPuc , thank you for your fast answer. 
The name in the Prusa Slicr looks like this:
[image]
But in Fusion Text is shown correctly. This happened  only in the new version in combination with Fusion. If I load an older project produced with the version before, the names are also already in clear text correct. When I add a new part, from Fusion the name is cryptic for this new part only. 
Also having the slicr closed and let it open by the Fusion export has the cryptic name result. I can rename it and than the  text will be correct shown and transferred to Prusa Connect and th MK4S. But if I didn't rename the part, the cryptic name will be send to prusa connect and to the printer. 
When I load STL files from the HDD instead from Fusion, the name is shown correct in the Slicr.
It seems only a topic of the interface between Fusion360 and Prusa Slicer making a direct transfer of the object, maybe only on MAC.
Is this now more clear? If not, please let me know.

TerPuc, 2024-12-20:
Dear @BernhardFe, thank you for your reply. I am afraid we will need more info. Could you provide us with video of the whole proces and/or a screenshot of the Fusion with the project? Thanks!

BernhardFe, 2024-12-20:
Dear @TerPuc, I did several tests with other Slicr projects and with other projects in Fusion in several combinations. In the end it happened only for one dedicated part in a Fusion project. Maybe it was a coincidence that it happened at the same time when I updated the slicr. In Fusion I made a copy of the object and deleted the former one with the bug. The copy was working correctly again and the correct name was shown in the Slicr. I was also not able to reproduce this again. No idea why this happened. Prusa Slicer is ok. I close the case now. Sorry taking your time and thanks for your effort.

closed 2024-12-20 by BernhardFe