prusa3d/PrusaSlicer #13798: 2.9.0 rct

eddie0258, 2024-12-19:
### Description of the bug

the program shuts down unexpectedly without notice. this occurs when I am working on painting a structure. it does it quite frequently. able to download object as STL using mmu3 and mk4s model. 

### Project file & How to reproduce

-----------
Full Report
-----------

{"app_name":"PrusaSlicer","timestamp":"2024-12-19 11:02:06.00 -0500","app_version":"PrusaSlicer PrusaSlicer-2.9.0-rc1","slice_uuid":"df42d0e9-65b6-343e-8b48-a5bd2969db55","build_version":"PrusaSlicer-2.9.0-rc1","platform":1,"bundleID":"com.prusa3d.slic3r\/","share_with_app_devs":1,"is_first_party":0,"bug_type":"309","os_version":"macOS 15.1.1 (24B91)","roots_installed":0,"name":"PrusaSlicer","incident_id":"C33E0C78-5A94-4B36-B179-5233111673A8"}
{
  "uptime" : 1000000,
  "procRole" : "Foreground",
  "version" : 2,
  "userID" : 501,
  "deployVersion" : 210,
  "modelCode" : "iMac19,1",
  "coalitionID" : 28751,
  "osVersion" : {
    "train" : "macOS 15.1.1",
    "build" : "24B91",
    "releaseType" : "User"
  },
  "captureTime" : "2024-12-19 11:01:49.1171 -0500",
  "codeSigningMonitor" : 0,
  "incident" : "C33E0C78-5A94-4B36-B179-5233111673A8",
  "pid" : 60139,
  "cpuType" : "X86-64",
  "roots_installed" : 0,
  "bug_type" : "309",
  "procLaunch" : "2024-12-18 13:07:14.0169 -0500",
  "procStartAbsTime" : 969910174435850,
  "procExitAbsTime" : 1042269503888665,
  "procName" : "PrusaSlicer",
  "procPath" : "\/Volumes\/VOLUME\/*\/PrusaSlicer.app\/Contents\/MacOS\/PrusaSlicer",
  "bundleInfo" : {"CFBundleShortVersionString":"PrusaSlicer PrusaSlicer-2.9.0-rc1","CFBundleVersion":"PrusaSlicer-2.9.0-rc1","CFBundleIdentifier":"com.prusa3d.slic3r\/"},
  "storeInfo" : {"deviceIdentifierForVendor":"E4CD4D9B-E54F-566E-B66A-845052FCDC99","thirdParty":true},
  "parentProc" : "launchd",
  "parentPid" : 1,
  "coalitionName" : "com.prusa3d.slic3r\/",
  "crashReporterKey" : "83E3C7DE-9CC1-530D-7C16-74742CBBF6C7",
  "codeSigningID" : "com.prusa3d.slic3r\/",
  "codeSigningTeamID" : "DKPB65N43Z",
  "codeSigningFlags" : 570495745,
  "codeSigningValidationCategory" : 6,
  "codeSigningTrustLevel" : 4294967295,
  "bootSessionUUID" : "0ADB638E-6246-41F4-8FC7-1D28F7792B93",
  "wakeTime" : 67512,
  "sleepWakeUUID" : "5104BCFE-4BB6-4B54-92EC-D64434A4EC81",
  "sip" : "enabled",
  "vmRegionInfo" : "0 is not in any region.  Bytes before following region: 4535648256\n      REGION TYPE                    START - END         [ VSIZE] PRT\/MAX SHRMOD  REGION DETAIL\n      UNUSED SPACE AT START\n--->  \n      __TEXT                      10e588000-111f94000    [ 58.0M] r-x\/r-x SM=COW  \/var\/folders\/*\/PrusaSlicer.app\/Contents\/MacOS\/PrusaSlicer",
  "exception" : {"codes":"0x0000000000000001, 0x0000000000000000","rawCodes":[1,0],"type":"EXC_BAD_ACCESS","signal":"SIGSEGV","subtype":"KERN_INVALID_ADDRESS at 0x0000000000000000"},
  "termination" : {"flags":0,"code":11,"namespace":"SIGNAL","indicator":"Segmentation fault: 11","byProc":"exc handler","byPid":60139},
  "vmregioninfo" : "0 is not in any region.  Bytes before following region: 4535648256\n      REGION TYPE                    START - END         [ VSIZE] PRT\/MAX SHRMOD  REGION DETAIL\n      UNUSED SPACE AT START\n--->  \n      __TEXT                      10e588000-111f94000    [ 58.0M] r-x\/r-x SM=COW  \/var\/folders\/*\/PrusaSlicer.app\/Contents\/MacOS\/PrusaSlicer",
  "extMods" : {"caller":{"thread_create":0,"thread_set_state":0,"task_for_pid":0},"system":{"thread_create":0,"thread_set_state":0,"task_for_pid":0},"targeted":{"thread_create":0,"thread_set_state":0,"task_for_pid":0},"warnings":0},
  "faultingThread" : 8,
  "threads" : [{"threadState":{"r13":{"value":0},"rax":{"value":4598297096,"symbolLocation":16,"symbol":"vtable for wxObject"},"rflags":{"value":514},"cpu":{"value":0},"r14":{"value":140442333316064},"rsi":{"value":4627571712},"r8":{"value":96},"cr2":{"value":0},"rdx":{"value":4386272},"r10":{"value":140441448456192},"r9":{"value":18446744073709547520},"r15":{"value":4},"rbx":{"value":6},"trap":{"value":222},"err":{"value":0},"r11":{"value":6},"rip":{"value":4586323840},"rbp":{"value":140701813101520},"rsp":{"value":140701813101400},"r12":{"value":140701813101616},"rcx":{"value":0},"flavor":"x86_THREAD_STATE","rdi":{"value":140701813101464}},"id":10068723,"name":"slic3r_main","queue":"com.apple.main-thread","frames":[{"imageOffset":50675584,"symbol":"wxObject::UnRef()","symbolLocation":0,"imageIndex":0},{"imageOffset":48037184,"symbol":"wxGCDCImpl::DoDrawBitmap(wxBitmap const&, int, int, bool)","symbolLocation":752,"imageIndex":0},{"imageOffset":18605215,"symbol":"TextInput::render(wxDC&)","symbolLocation":255,"imageIndex":0},{"imageOffset":18597624,"symbol":"TextInput::paintEvent(wxPaintEvent&)","symbolLocation":40,"imageIndex":0},{"imageOffset":50974655,"symbol":"wxEventHashTable::HandleEvent(wxEvent&, wxEvtHandler*)","symbolLocation":479,"imageIndex":0},{"imageOffset":50980099,"symbol":"wxEvtHandler::ProcessEvent(wxEvent&)","symbolLocation":211,"imageIndex":0},{"imageOffset":50980433,"symbol":"wxEvtHandler::ProcessEventLocally(wxEvent&)","symbolLocation":161,"imageIndex":0},{"imageOffset":50979929,"symbol":"wxEvtHandler::ProcessEvent(wxEvent&)","symbolLocation":41,"imageIndex":0},{"imageOffset":50981804,"symbol":"wxEvtHandler::SafelyProcessEvent(wxEvent&)","symbolLocation":12,"imageIndex":0},{"imageOffset":49376843,"symbol":"wxWindow::MacDoRedraw(long)","symbolLocation":683,"imageIndex":0},{"imageOffset":50010411,"symbol":"wxWidgetCocoaImpl::drawRect(void*, NSView*, void*)","symbolLocation":731,"imageIndex":0},{"imageOffset":49997782,"symbol":"wxOSX_drawRect(NSView*, objc_selector*, CGRect)","symbolLocation":86,"imageIndex":0},{"imageOffset":1587039,"symbol":"_NSViewDrawRect","symbolLocation":142,"imageIndex":5},{"imageOffset":12775631,"symbol":"-[NSView _recursive:displayRectIgnoringOpacity:inContext:stopAtLayerBackedViews:]","symbolLocation":1891,"imageIndex":5},{"imageOffset":12776485,"symbol":"-[NSView _recursive:displayRectIgnoringOpacity:inContext:stopAtLayerBackedViews:]","symbolLocation":2745,"imageIndex":5},{"imageOffset":1585169,"symbol":"-[NSView(NSLayerKitGlue) _drawViewBackingLayer:inContext:drawingHandler:]","symbolLocation":762,"imageIndex":5},{"imageOffset":8803578,"symbol":"-[NSViewBackingLayer drawInContext:]","symbolLocation":64,"imageIndex":5},{"imageOffset":4220401,"imageIndex":5},{"imageOffset":4952400,"imageIndex":5},{"imageOffset":4933469,"imageIndex":5},{"imageOffset":4951120,"imageIndex":5},{"imageOffset":4941222,"imageIndex":5},{"imageOffset":5360233,"imageIndex":5},{"imageOffset":5361723,"imageIndex":5},{"imageOffset":8802588,"symbol":"-[NSViewBackingLayer display]","symbolLocation":1763,"imageIndex":5},{"imageOffset":136076,"symbol":"CA::Layer::display_if_needed(CA::Transaction*)","symbolLocation":882,"imageIndex":6},{"imageOffset":1847804,"symbol":"CA::Context::commit_transaction(CA::Transaction*, double, double*)","symbolLocation":766,"imageIndex":6},{"imageOffset":8071,"symbol":"CA::Transaction::commit()","symbolLocation":725,"imageIndex":6},{"imageOffset":1655105,"symbol":"__62+[CATransaction(NSCATransaction) NS_setFlushesWithDisplayLink]_block_invoke","symbolLocation":289,"imageIndex":5},{"imageOffset":13083447,"symbol":"___NSRunLoopObserverCreateWithHandler_block_invoke","symbolLocation":41,"imageIndex":5},{"imageOffset":505235,"symbol":"__CFRUNLOOP_IS_CALLING_OUT_TO_AN_OBSERVER_CALLBACK_FUNCTION__","symbolLocation":23,"imageIndex":7},{"imageOffset":505006,"symbol":"__CFRunLoopDoObservers","symbolLocation":503,"imageIndex":7},{"imageOffset":502343,"symbol":"__CFRunLoopRun","symbolLocation":905,"imageIndex":7},{"imageOffset":499854,"symbol":"CFRunLoopRunSpecific","symbolLocation":550,"imageIndex":7},{"imageOffset":982723,"symbol":"RunCurrentEventLoopInMode","symbolLocation":292,"imageIndex":8},{"imageOffset":1005287,"symbol":"ReceiveNextEventCommon","symbolLocation":201,"imageIndex":8},{"imageOffset":1006146,"symbol":"_BlockUntilNextEventMatchingListInModeWithFilter","symbolLocation":66,"imageIndex":8},{"imageOffset":260991,"symbol":"_DPSNextEvent","symbolLocation":902,"imageIndex":5},{"imageOffset":10962612,"symbol":"-[NSApplication(NSEventRouting) _nextEventMatchingEventMask:untilDate:inMode:dequeue:]","symbolLocation":1290,"imageIndex":5},{"imageOffset":199769,"symbol":"-[NSApplication run]","symbolLocation":610,"imageIndex":5},{"imageOffset":49876250,"symbol":"wxGUIEventLoop::OSXDoRun()","symbolLocation":186,"imageIndex":0},{"imageOffset":51026253,"symbol":"wxCFEventLoop: [... cut ...]
    "path" : "\/Volumes\/VOLUME\/*\/PrusaSlicer.app\/Contents\/MacOS\/PrusaSlicer",
    "name" : "PrusaSlicer",
    "CFBundleVersion" : "PrusaSlicer-2.9.0-rc1"
[... 194 lines cut ...]
  ],
  "experiments" : [
    {
      "treatmentId" : "45f4e2a5-551b-4bc2-a2dc-19c244dda8f8",
      "experimentId" : "6643969b3099cf28e049862f",
      "deploymentId" : 400000007
    }
  

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.9.0 act

### Operating system

Mac OS Sequoia 15.1.1

### Printer model

MK4S with MMU3

lukasmatena, 2024-12-19:
Thanks for the report.
Can you please share your 3MF project?
Are you able to reproduce this issue in PrusaSlicer 2.8.1?

lukasmatena, 2024-12-20:
We are struggling to reproduce the issue. Were you doing something specific before the crash, or did you just open PS, open model and start painting? In particular, were you switching printer profiles or adjusting number of used extruders anytime during that PrusaSlicer session (i.e. not necessarily immediately before the crash)?

lukasmatena, 2024-12-20:
We have fixed an issue which could have caused this, but given the amount of info we got, it is not 100% clear that it was the same issue as yours. Could you please retest it in [2.9.0](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.9.0). Thanks.

eddie0258, 2024-12-20:
I downloaded a stl and started to paint it and each time it shut down. It was with this specific stl.
Eddie Acosta

lukasmatena, 2024-12-20:
Thanks for the update. I am afraid that we are out of the ideas, at least until we have an STL to reproduce the problem.

closed 2026-09-29 by prusaslicer-bot[bot]