prusa3d/PrusaSlicer #13801: Can't connect to printer — Error 7

mix579, 2024-12-19:
### Description of the bug

With the recent RC I'm getting the following error when trying to upload a file to my Prusa Mini+:

Upload has failed. There's no suitable storage at xxxxx. 
Couldn't connect to server:
Error 7

I can go to the printer in Connect or on Prusaslicer's Physical Printer tab and can browse the USB stick no problem. The Test button in the Physical Printer configuration screen also works fine.

### Project file & How to reproduce

Sorry, proprietary model. Easy to reproduce. Slice, click on the upload button.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.9.0 rc1

### Operating system

Mac OS 15.1.1

### Printer model

Prusa Mini+

fitrider82, 2025-01-08:
I too, am suddenly running into this issue. My Octoprint and MK3S+ have been working flawlessly for years. I've tried removing and reconfiguring the printer through PrusaSlicer, restarting my network, rebooting the Pi. I can ping the Pi's IP address without issue and the web UI is working flawlessly.

Carsten3P, 2025-01-11:
I have exactly the same issue, neither dns nor ip-address works.
Prusaslicer 2.9.0, Octoprint  1.10.3.
Any Idea how to get this working again?

richardchips, 2025-01-11:
Same here.  Tried reverting back to Prusaslicer 2.8.1, but no joy.  Latest Octoprint 1.10.3.  Have similarly tried all steps listed above, but no joy.

Further: I can't spot any failed connection attempts in the Octoprint logs.

richardchips, 2025-01-11:
Further diagnostics - still without success:

 - Fresh install of Prusaslicer 2.7.4 (after 2.8 failed similarly) - including delete of previous config
 - Installed fresh image of Octopi with earlier version of Octoprint 1.9.3

 - From Prusaslicer laptop (Mac) terminal, running **curl -H "X-Api-Key: <redacted>" http://octopi3** gives two things:
      1) a response (lots of HTML) to the curl in the terminal
      2) a log entry in Octoprint's octoprint.log:

2025-01-11 18:12:42,675 - octoprint.server.util.flask - INFO - Passively logging in user pi from <redacted>
2025-01-11 18:12:42,677 - octoprint.access.users - INFO - Logged in user: pi

 - This is in contrast to behaviour when I hit the "Test" button in Prusaslicer 2.7.4 (or 2.8 or 2.9):
      1) Error in Prusaslicer: Failed to connect to octopi3 port 80 after 16 ms: Couldn't connect to server [Error 7]
      2) No corresponding log entry on Octoprint log; log is silent.

Rather strange, given downgrading of Prusaslicer and fresh installing of Octoprint.

Will now stop hijacking your thread, sorry!

(Did you have any success in resolving, @mix579?)

Carsten3P, 2025-01-12:
Thanks a lot  @richardchips, got it working again with this tip. (also MacOS)
My way without reinstallation: 

1. Start Prusaslicer -> Test Connection (Error 7)
2. Quit Prusaslicer, goto Settings-Security-Local Network - switch Psusaslicer OFF!
3. Start Prusasliver Again -> Test Connection (Error 7) 
4. Quit Prusaslicer, goto Settings-Security-Local Network - switch Psusaslicer ON
5. Start Prusaslicer Again an Test Works ok :-)

closed 2026-09-29 by prusaslicer-bot[bot]