prusa3d/PrusaSlicer #13804: Several issues with text editing / placement

wlcina, 2024-12-20:
### Description of the bug

There are several issues with text editing including latest beta 2.90, or the 2.8.1 version. 

1) clicking on gear and Edit text - there is a big chance the window for editing text (where I can change font and other parameters) do not appear. It just do nothing. Upon closing prusaslicer and opening project again, the editing text window appears correctly (for a while)

2) When editing text (meaning changing / retyping text field) - causes world size parameters to infinity 
[image]
[testissues.zip](https://github.com/user-attachments/files/18207690/testissues.zip)
3) Position of text in project changes randomly, I want to have text at the top of the existing stl, but I have to play with the position coordinates to properly place text at the bottom layer or the top layer. Coordinates changes randomly in all objects within the group (often sinking etc). I suggest to not to force changing coordinates, but keep the user defined coordinates permanently rather than helping him with software, it causes a lot of mess. 

### Project file & How to reproduce

Open file, try edit text (any of 2 text), and type any letter (eg just add letter "a")

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.8.1, 2.9.0

### Operating system

Windows 11

### Printer model

XL

wlcina, 2024-12-20:
Found the issue for infinity size,  slicer itself changes the thickness to 0 for some reason (in the text parameter windows)

wlcina, 2024-12-20:
Found also when the change is happening. Was working with the file  for minutes without an issue (also copy and paste without any issue) - as soon as I saved the file (CTRL S) then suddenly issues appeared - meaning:
- thickness of text was set to zero (thus inf error)
- window for text parameter did not appear anymore, need to restart slicer to get into it

Xargas, 2025-02-15:
I can confirm point (3) of the topic: when text is edited, the placement seems to "change randomly".
I used the model given in the ZIP file above, but also observed this, when printing my model https://www.printables.com/model/1190147-gridfinity-prusa-hotend-holder and changing texts. I always had to manually re-adjust the position, which was already correct before the automatic position change of the slicer!

So I *strongly* vote for a button to perform the "automatic alignment helper tool" and otherwise stick with the position that is specified when the text is changed.

closed 2026-09-29 by prusaslicer-bot[bot]