prusa3d/PrusaSlicer #13806: First Layer Travel Speed

Hirumaxim, 2024-12-20:
PrusaSlicer Does not have the function to change the travel speed at the first layer. its much needed in situation like this where there are small text. I tried numerous methods to limit the travel speed in the slicer but i was not able to do it. 
[image]

emelymw, 2024-12-20:
hey, 
the prusaslicer has a setting option for the first layer. 
All you have to do is go to ‘Normal or Expert mode’ and then under Print Settings < Speed < Modifiers < First layer speed. There you can adapt the speed for the first layer.

Hirumaxim, 2024-12-20:
[image]

This is default setting with IS in Prusa Mini. This is changing the speed to 1%

[image]

Still the issue is there? Any Fixes

Thanks

Hirumaxim, 2024-12-20:
I just want the No_Print Travel to be lower than what is there. If it reach 400mm per second. It will rip off PLA from the Smooth PEI Sheet

emelymw, 2024-12-20:
what if you use the setting "first layer" under acceleration control (advanced) to control this ? Is this working ?

Hirumaxim, 2024-12-20:
[image]
This one right
[image]
Still Same Issue

Hirumaxim, 2024-12-20:
[image]

This is a G-Code Generated by Cura for Prusa Mini Profile (Without IS)

In there, there is setting called "Initial Layer Travel Speed"

This command is able to override the Non_Print Travel Speed of 400mm/s to 20mm/s as i have set 

[image]

Why we cant do this in Prusaslicer?

closed 2026-09-29 by prusaslicer-bot[bot]