prusa3d/PrusaSlicer #13809: ColorFabb VarioShore TPU Filament preset unavailable ?

rcmz, 2024-12-20:
### Description of the bug

ColorFabb VarioShore TPU is not listed in the filament presets wizard
[image]
Yet when I tried to create a custom profile for it, it tells me that a system profile with this name already exists
[image]
And indeed this profile exists in the source code
https://github.com/prusa3d/PrusaSlicer/blob/5c7888b11453eb96a95b49dfd4cbd3b56db64eb8/resources/profiles/Templates.ini#L2140
[image]
So why is it not available in the filament presets wizard ??

### Project file & How to reproduce

[Untitled.zip](https://github.com/user-attachments/files/18213996/Untitled.zip)

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.8.1 Flatpak, 2.9 Windows

### Operating system

Arch Linux

### Printer model

Prusa MINI

dusanmerta, 2025-05-12:
I am experiencing the same issue.

PrusaSlicer 2.9.2
MacOS 14.7.5.

rcmz, 2025-05-12:
Oh yes indeed.
While I understand where this restriction comes from, I feel like it is more frustrating than useful. And the fact of the matter is that the Prusa MINI can print TPUs / VarioShore, even if it is not great at it. IMO putting a clear but non-blocking warning is always better than a hard restriction.

closed 2026-09-29 by prusaslicer-bot[bot]