prusa3d/PrusaSlicer #13813: slic3rutils_tests segfaults

cryptomilk, 2024-12-20:
### Description of the bug

I'm trying to package PrusaSlicer 2.9.0 for openSUSE Tumbleweed. However with 2.9.0 slic3rutils_tests

```
(gdb) r                                                                                                                                                                                       
Starting program: /home/abuild/rpmbuild/BUILD/PrusaSlicer-version_2.9.0/build/tests/slic3rutils/slic3rutils_tests exclude:\[NotWorking\] -s
[...]
Program received signal SIGSEGV, Segmentation fault.

(gdb) bt
#0  Slic3r::ConfigOptionEnumDef::finalize_closed_enum (this=0x555556d34840) at /home/abuild/rpmbuild/BUILD/PrusaSlicer-version_2.9.0/src/libslic3r/../libslic3r/Config.hpp:2195
#1  Slic3r::ConfigDef::finalize (this=0x555556c8fe00 <Slic3r::print_config_def>) at /home/abuild/rpmbuild/BUILD/PrusaSlicer-version_2.9.0/src/libslic3r/Config.cpp:342
#2  0x00007ffff6ef39b8 in __static_initialization_and_destruction_0 () at /home/abuild/rpmbuild/BUILD/PrusaSlicer-version_2.9.0/src/libslic3r/PrintConfig.cpp:5069
#3  0x00007ffff6f2531a in _sub_I_65535_0.0 () at /home/abuild/rpmbuild/BUILD/PrusaSlicer-version_2.9.0/build/src/slic3r-arrange/libslic3r-arrange.so
#4  0x00007ffff7fc969e in call_init () at /lib64/ld-linux-x86-64.so.2
#5  0x00007ffff7fc979c in _dl_init () at /lib64/ld-linux-x86-64.so.2
#6  0x00007ffff7fe2620 in _dl_start_user () at /lib64/ld-linux-x86-64.so.2
#7  0x0000000000000003 in ??? ()
#8  0x00007fffffffeb1c in ??? ()
#9  0x00007fffffffeb7c in ??? ()
#10 0x00007fffffffeb91 in ??? ()
#11 0x0000000000000000 in ??? ()
(gdb) p enum_int
$1 = (std::tuple_element<1, std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, int> const>::type &) <error reading variable: Cannot access memor
y at address 0x40>
(gdb) p enum_name
$2 = (std::tuple_element<0, std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, int> const>::type &) <error reading variable: Cannot access memor
y at address 0x20>
(gdb) p *m_enum_keys_map
$3 = {_M_t = {
    _M_impl = {<std::allocator<std::_Rb_tree_node<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, int> > >> = {<std::__new_allocator<std::_Rb
_tree_node<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, int> > >> = {<No data fields>}, <No data fields>}, <std::_Rb_tree_key_compare<std:
:less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >> = {
        _M_key_compare = {<std::binary_function<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std:
:allocator<char> >, bool>> = {<No data fields>}, <No data fields>}}, <std::_Rb_tree_header> = {_M_header = {_M_color = std::_S_red, _M_parent = 0x0, _M_left = 0x0, _M_right = 0x0}, 
        _M_node_count = 0}, <No data fields>}}}
```

I'm a C guy and not a C++ guy. I've never seen such a for loop before. I would need some help to debug this further and understand what is going on.

### Project file & How to reproduce

See https://build.opensuse.org/package/show/home:gladiac:branches:science/PrusaSlicer

Can be built locally with `osc build` and you can go to the chroot with `osc chroot` to rerun the test and attach a debugger.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.9.0

### Operating system

openSUSE Tumbleweed

### Printer model

Chameleon

cryptomilk, 2024-12-21:
Unfortunately AddressSanitizer can not provide additional info :-(

```
6: AddressSanitizer:DEADLYSIGNAL
6: =================================================================
6: ==734335==ERROR: AddressSanitizer: SEGV on unknown address 0x000000000040 (pc 0x000002617fc7 bp 0x7ffe89861630 sp 0x7ffe89861460 T0)
6: ==734335==The signal is caused by a READ memory access.
6: ==734335==Hint: address points to the zero page.
6:     #0 0x2617fc7 in Slic3r::ConfigOptionEnumDef::finalize_closed_enum() /home/asn/workspace/pkg/obs/PrusaSlicer-version_2.9.0/src/libslic3r/../libslic3r/Config.hpp:2195
6:     #1 0x2617fc7 in Slic3r::ConfigDef::finalize() /home/asn/workspace/pkg/obs/PrusaSlicer-version_2.9.0/src/libslic3r/Config.cpp:342
6:     #2 0x30e0b90 in Slic3r::PrintConfigDef::PrintConfigDef() /home/asn/workspace/pkg/obs/PrusaSlicer-version_2.9.0/src/libslic3r/PrintConfig.cpp:346
6:     #3 0x7fa5da5e7fe2 in __static_initialization_and_destruction_0 /home/asn/workspace/pkg/obs/PrusaSlicer-version_2.9.0/src/libslic3r/PrintConfig.cpp:5069
6:     #4 0x7fa5da7849be in _sub_I_65535_0.0 (/home/asn/workspace/pkg/obs/PrusaSlicer-version_2.9.0/obj/src/slic3r-arrange/libslic3r-arrange.so+0x3849be) (BuildId: 96c246bc87a7f5d611ce253ffdee45c3584aab53)
6:     #5 0x7fa5dfc1069d in call_init /usr/src/debug/glibc-2.40/elf/dl-init.c:74
6:     #6 0x7fa5dfc1069d in call_init /usr/src/debug/glibc-2.40/elf/dl-init.c:26
6:     #7 0x7fa5dfc1079b in _dl_init /usr/src/debug/glibc-2.40/elf/dl-init.c:121
6:     #8 0x7fa5dfc2961f  (/lib64/ld-linux-x86-64.so.2+0x1d61f) (BuildId: 8827ac3d78050dde04b918bd2fb81eb9aba67664)
6: 
6: AddressSanitizer can not provide additional info.
```

fordfrog, 2024-12-21:
we have the same issue in gentoo... i'm just recompiling with debug to see more...

cryptomilk, 2024-12-21:
The way they implemented asan support in cmake is strange. Normally you would do -DCMAKE_BUILD_TYPE=AddressSanitizer. In case someone from Prusa is listening, take a look how I do it here: https://gitlab.com/cmocka/cmocka/-/blob/master/cmake/Modules/DefineCompilerFlags.cmake?ref_type=heads

fordfrog, 2024-12-21:
@cryptomilk my output is a bit different from yours when it comes to the variables:
```
Program received signal SIGSEGV, Segmentation fault.
Slic3r::ConfigOptionEnumDef::finalize_closed_enum (this=0x5555576bf2a0) at /usr/src/debug/media-gfx/prusaslicer-9999/prusaslicer-9999/src/libslic3r/Config.hpp:2195
2195	            if (enum_int >= int(this->values().size()) || this->value(enum_int) != enum_name) {
(gdb) bt
#0  Slic3r::ConfigOptionEnumDef::finalize_closed_enum (this=0x5555576bf2a0) at /usr/src/debug/media-gfx/prusaslicer-9999/prusaslicer-9999/src/libslic3r/Config.hpp:2195
#1  Slic3r::ConfigDef::finalize (this=this@entry=0x5555575bfc40 <Slic3r::print_config_def>) at /usr/src/debug/media-gfx/prusaslicer-9999/prusaslicer-9999/src/libslic3r/Config.cpp:342
#2  0x0000555555e9ed75 in Slic3r::PrintConfigDef::PrintConfigDef (this=0x5555575bfc40 <Slic3r::print_config_def>) at /usr/src/debug/media-gfx/prusaslicer-9999/prusaslicer-9999/src/libslic3r/PrintConfig.cpp:346
#3  0x00007ffff6d752ab in __static_initialization_and_destruction_0 () at /usr/src/debug/media-gfx/prusaslicer-9999/prusaslicer-9999/src/libslic3r/PrintConfig.cpp:5069
#4  0x00007ffff7fca54e in call_init () at /lib64/ld-linux-x86-64.so.2
#5  0x00007ffff7fca63c in _dl_init () at /lib64/ld-linux-x86-64.so.2
#6  0x00007ffff7fe2920 in _dl_start_user () at /lib64/ld-linux-x86-64.so.2
#7  0x0000000000000001 in ??? ()
#8  0x00007fffffffd63f in ??? ()
#9  0x0000000000000000 in ??? ()
(gdb) print enum_int
$1 = (std::tuple_element<1, std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, int> const>::type &) <error reading variable: Cannot access memory at address 0x40>
(gdb) print enum_name
$2 = <error reading variable: Cannot access memory at address 0x28>
(gdb) print *m_enum_keys_map
$3 = std::map with 0 elements
```

fordfrog, 2024-12-21:
so after some debugging, it looks like `set_enum_map()` is called many times before `finalize_closed_enum` is called, but still `m_enum_keys_map` is empty:
```
Reading symbols from prusa-slicer...
Reading symbols from /usr/lib/debug//usr/bin/prusa-slicer.debug...
(gdb) break ConfigOptionEnumDef::finalize_closed_enum
quit
Breakpoint 1 at 0x779868: file /usr/src/debug/media-gfx/prusaslicer-9999/prusaslicer-9999/src/libslic3r/Config.hpp, line 2192.
(gdb) break ConfigOptionEnumDef::set_enum_map

Breakpoint 2 at 0x929f21: ConfigOptionEnumDef::set_enum_map. (34 locations)
(gdb) run

Starting program: /usr/bin/prusa-slicer 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib64/libthread_db.so.1".

Breakpoint 2.2, Slic3r::ConfigOptionEnumDef::set_enum_map<Slic3r::PrinterTechnology> (this=0x5555576bb500) at /usr/src/debug/media-gfx/prusaslicer-9999/prusaslicer-9999/src/libslic3r/Config.hpp:2209
2209	    void set_enum_map()

(gdb) continue

Continuing.

Breakpoint 2.3, Slic3r::ConfigOptionEnumDef::set_enum_map<Slic3r::GCodeThumbnailsFormat> (this=0x5555576bc680) at /usr/src/debug/media-gfx/prusaslicer-9999/prusaslicer-9999/src/libslic3r/Config.hpp:2209
2209	    void set_enum_map()

(gdb) continue

Continuing.

Breakpoint 2.4, Slic3r::ConfigOptionEnumDef::set_enum_map<Slic3r::AuthorizationType> (this=0x5555576be7e0) at /usr/src/debug/media-gfx/prusaslicer-9999/prusaslicer-9999/src/libslic3r/Config.hpp:2209
2209	    void set_enum_map()

(gdb) continue

Continuing.

Breakpoint 2.14, Slic3r::ConfigOptionEnumDef::set_enum_map<Slic3r::ArcFittingType> (this=0x5555576bf2a0) at /usr/src/debug/media-gfx/prusaslicer-9999/prusaslicer-9999/src/libslic3r/Config.hpp:2209
2209	    void set_enum_map()

(gdb) continue

Continuing.

Breakpoint 2.15, Slic3r::ConfigOptionEnumDef::set_enum_map<Slic3r::TopOnePerimeterType> (this=0x5555576c34a0) at /usr/src/debug/media-gfx/prusaslicer-9999/prusaslicer-9999/src/libslic3r/Config.hpp:2209
2209	    void set_enum_map()

(gdb) continue

Continuing.

Breakpoint 2.16, Slic3r::ConfigOptionEnumDef::set_enum_map<Slic3r::BrimType> (this=0x5555576c6b90) at /usr/src/debug/media-gfx/prusaslicer-9999/prusaslicer-9999/src/libslic3r/Config.hpp:2209
2209	    void set_enum_map()

(gdb) continue

Continuing.

Breakpoint 2.17, Slic3r::ConfigOptionEnumDef::set_enum_map<Slic3r::EnsureVerticalShellThickness> (this=0x5555576cb3f0) at /usr/src/debug/media-gfx/prusaslicer-9999/prusaslicer-9999/src/libslic3r/Config.hpp:2209
2209	    void set_enum_map()

(gdb) continue

Continuing.

Breakpoint 2.18, Slic3r::ConfigOptionEnumDef::set_enum_map<Slic3r::InfillPattern> (this=0x5555576cb8c0) at /usr/src/debug/media-gfx/prusaslicer-9999/prusaslicer-9999/src/libslic3r/Config.hpp:2209
2209	    void set_enum_map()

(gdb) continue

[... 162 lines cut ...]

Breakpoint 2.8, Slic3r::ConfigOptionEnumDef::set_enum_map<Slic3r::TiltSpeeds> (this=0x5555577287e0) at /usr/src/debug/media-gfx/prusaslicer-9999/prusaslicer-9999/src/libslic3r/Config.hpp:2209
2209	    void set_enum_map()

(gdb) continue

Continuing.

Breakpoint 2.9, Slic3r::ConfigOptionEnumDef::set_enum_map<Slic3r::TiltSpeeds> (this=0x555557728f10) at /usr/src/debug/media-gfx/prusaslicer-9999/prusaslicer-9999/src/libslic3r/Config.hpp:2209
2209	    void set_enum_map()

(gdb) continue

Continuing.

Breakpoint 1.1, Slic3r::ConfigOptionEnumDef::finalize_closed_enum (this=0x5555576bf2a0) at /usr/src/debug/media-gfx/prusaslicer-9999/prusaslicer-9999/src/libslic3r/Config.hpp:2192
2192	        m_values_ordinary = true;

(gdb) print m_enum_keys_map

$1 = (const Slic3r::t_config_enum_values *) 0x5555575c0a80 <Slic3r::s_keys_map_ArcFittingType>
(gdb) print *m_enum_keys_map

$2 = std::map with 0 elements
```

cryptomilk, 2024-12-21:
I've started a git bisect. 

* The commit 3e33631abf384895c989b33f723885a4f1cb52f8 doesn't compile, so I'm not able to run the test with it. Each commit should compile and work especially that you can do git bisect. However I've marked it as "good".
* The next commit is 5199d8fb48b391b4cf268e74d78beba8db8e15e0 and the test fails.

This means either 3e33631abf384895c989b33f723885a4f1cb52f8 or 5199d8fb48b391b4cf268e74d78beba8db8e15e0 introduced the issue.

cryptomilk, 2024-12-21:
@SachCZ ^^

l29ah, 2024-12-24:
It crashes like that in `prusa-slicer` proper for me right at the start on master on Gentoo Lignux.

cryptomilk, 2025-01-02:
What I understand, but I'm not a C++ expert. We are in the calling library constructors and there is one for the static global variable: `const PrintConfigDef print_config_def;`.

For this we call the class constructor `PrintConfigDef::PrintConfigDef()`. It looks like something goes wrong in this function. As we segfault in the lass call to `this->finalize()`. This means one of the calls before is either doing something wrong, or we miss to call something required.

fordfrog, 2025-01-04:
[this](https://github.com/prusa3d/PrusaSlicer/pull/13896) resolved the segfault for us in gentoo.

cryptomilk, 2025-01-04:
Yeah, I guess that some initializer is not called, so it isn't fully set up yet. Linking statically solves that.

hyperair, 2025-01-07:
Thanks @fordfrog, that fixed the issues I had packaging this for Debian.

closed 2025-04-23 by cryptomilk