prusa3d/PrusaSlicer #13814: Multi-bed numbers interfere with objects

Bmorr1123, 2024-12-20:
### Description of the bug

I have a few things being printed in the bottom left of the bed, and the numbers prevent part of them from being sliced. I have attached screenshots of the objects, and of the first layers.
[image]
[image]
[image]
[image]
[image]
[image]

2 of the images are from another user who recreated it on their system.

### Project file & How to reproduce

[Cincinnati Sports Coasters v26.zip](https://github.com/user-attachments/files/18215550/Cincinnati.Sports.Coasters.v26.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.9

### Operating system

Windows 11

### Printer model

MK4 and I am using virtual nozzles in slicer

me25542, 2024-12-21:
Can confirm. The issue only appears when using multiple build plates. I am running Windows 11 24h2, PrusaSlicer 2.9.0

[image]
[image]
[image]

[3mf.zip](https://github.com/user-attachments/files/18217375/bug.zip)

lukasmatena, 2024-12-22:
Thanks for the report. This unfortunately slipped through our testing. It is a visual glitch, slicing itself is not affected. It will be fixed in the next version.
SPE-2632

lukasmatena, 2025-02-18:
Should be fixed in [PrusaSlicer 2.9.1-alpha1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.9.1-alpha1). Closing.

closed 2025-02-18 by lukasmatena