prusa3d/PrusaSlicer #13815: X-Y hole compensation feature

NOSSUDAK, 2024-12-20:
**Is your feature request related to a problem? Please describe.**
 I'm always frustrated when printed part dimensions are not accurate

**Describe the solution you'd like**
Add the X-Y hole compensation feature, similar to one in OrcaSlicer

**Describe how it would work**
I should be able to provide a compensation value to adjust the hole size

neophyl, 2024-12-21:
Please search for existing issues before opening new ones.  This is a duplicate of so many issues. Earliest being https://github.com/prusa3d/PrusaSlicer/issues/515 I think.
Or https://github.com/prusa3d/PrusaSlicer/issues/4561 or....

Duplicates don't get things fixed quicker, all they do is clutter up github

closed 2024-12-23 by NOSSUDAK

kubispe1, 2025-01-07:
Thanks for the feature request. This is a duplicate issue, so I am closing it.
At the moment, we don’t have it directly on our plan, but we are tracking the number of users requesting this functionality.