prusa3d/PrusaSlicer #13816: Crash on Text Modifier

dustjackettech, 2024-12-21:
### Description of the bug

When trying to add text to a surface and adjusting it to be a modifier that slightly sticks out from the surface, PrusaSlicer crashes.  I've tried rebooting and reinstalling 2.9 several times.  Still crashes.

### Project file & How to reproduce

[image]

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.9.0

### Operating system

macOS 15.2

### Printer model

XL 5T

dustjackettech, 2024-12-21:
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               PrusaSlicer [2946]
Path:                  /Applications/Original Prusa Drivers/PrusaSlicer.app/Contents/MacOS/PrusaSlicer
Identifier:            com.prusa3d.slic3r/
Version:               PrusaSlicer PrusaSlicer-2.9.0 (PrusaSlicer-2.9.0)
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]
User ID:               502

Date/Time:             2024-12-20 21:16:23.8792 -0600
OS Version:            macOS 15.2 (24C101)
Report Version:        12
Anonymous UUID:        8DE9631F-3F6B-A56A-517E-956EE13EADA9

Sleep/Wake UUID:       99C96308-0306-4C9E-BD9C-EC2E6AC7D8D8

Time Awake Since Boot: 3400 seconds

System Integrity Protection: enabled

Crashed Thread:        3

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x000000016df4c000
Exception Codes:       0x0000000000000001, 0x000000016df4c000

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [2946]

VM Region Info: 0x16df4c000 is not in any region.  Bytes after previous region: 1  Bytes before following region: 2293760
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      Stack                       16db44000-16df4c000    [ 4128K] rw-/rwx SM=PRV  thread 3
--->  GAP OF 0x230000 BYTES
      Stack Guard                 16e17c000-16e180000    [   16K] ---/rwx SM=NUL  

Thread 0:: slic3r_main Dispatch queue: com.Metal.CommandQueueDispatch
0   libsystem_kernel.dylib        	       0x195aaaf54 mach_msg2_trap + 8
1   libsystem_kernel.dylib        	       0x195abd604 mach_msg2_internal + 80
2   IOKit                         	       0x1995b51a0 io_connect_method + 524
3   IOKit                         	       0x1995b4f2c IOConnectCallMethod + 176
4   IOGPU                         	       0x1b6ad8d00 IOGPUCommandQueueSubmitCommandBuffers + 236
5   IOGPU                         	       0x1b6ac54ac -[IOGPUMetalCommandQueue _submitCommandBuffers:count:] + 356
6   IOGPU                         	       0x1b6ac531c -[IOGPUMetalCommandQueue submitCommandBuffers:count:] + 72
7   Metal                         	       0x1a09a61bc -[_MTLCommandQueue _submitAvailableCommandBuffers] + 488
8   Metal                         	       0x1a09ad050 __40-[_MTLCommandQueue submitCommandBuffer:]_block_invoke + 28
9   libdispatch.dylib             	       0x1959395b4 _dispatch_client_callout + 20
10  libdispatch.dylib             	       0x195948e08 _dispatch_lane_barrier_sync_invoke_and_complete + 56
11  Metal                         	       0x1a09ad00c -[_MTLCommandQueue submitCommandBuffer:] + 140
12  AppleMetalOpenGLRenderer      	       0x12842e0d4 GLDContextRec::flushContext(bool) + 624
13  GLEngine                      	       0x1f2c1f900 gleUnbindBufferObject + 68
14  GLEngine                      	       0x1f2c34bf8 gleUnbindDeleteHashNameAndObject + 84
15  GLEngine                      	       0x1f2b88d54 glDeleteBuffers_Exec + 2392
16  PrusaSlicer                   	       0x103297050 Slic3r::GUI::GLModel::reset() + 56
17  PrusaSlicer                   	       0x103198128 Slic3r::GUI::GLModel::~GLModel() + 36
18  PrusaSlicer                   	       0x1032a18b4 Slic3r::GUI::GLTexture::render_sub_texture(unsigned int, float, float, float, float, Slic3r::GUI::GLTexture::Quad_UVs const&) + 932
19  PrusaSlicer                   	       0x1032a736c Slic3r::GUI::GLToolbar::render_background(float, float, float, float, float, float) const + 636
20  PrusaSlicer                   	       0x1032a5cd0 Slic3r::GUI::GLToolbar::render_horizontal(Slic3r::GUI::GLCanvas3D const&) + 280
21  PrusaSlicer                   	       0x103196168 Slic3r::GUI::GLCanvas3D::_render_view_toolbar() const + 168
22  PrusaSlicer                   	       0x10317e7e8 Slic3r::GUI::GLCanvas3D::_render_overlays() + 128
23  PrusaSlicer                   	       0x10317a72c Slic3r::GUI::GLCanvas3D::render() + 1720
24  PrusaSlicer                   	       0x103187c1c Slic3r::GUI::GLCanvas3D::on_mouse(wxMouseEvent&) + 356
25  PrusaSlicer                   	       0x1050fecac wxEvtHandler::SearchDynamicEventTable(wxEvent&) + 360
26  PrusaSlicer                   	       0x1050fe920 wxEvtHandler::ProcessEventLocally(wxEvent&) + 88
27  PrusaSlicer                   	       0x1050fe6e8 wxEvtHandler::ProcessEvent(wxEvent&) + 56
28  PrusaSlicer                   	       0x1050fef84 wxEvtHandler::SafelyProcessEvent(wxEvent&) + 24
29  PrusaSlicer                   	       0x10502566c wxWidgetCocoaImpl::DoHandleMouseEvent(NSEvent*) + 92
30  PrusaSlicer                   	       0x10501e59c wxWidgetCocoaImpl::mouseEvent(NSEvent*, NSView*, void*) + 176
[... 543 lines cut ...]
}
}

Model: Mac15,11, BootROM 11881.61.3, proc 14:10:4 processors, 36 GB, SMC 
Graphics: Apple M3 Max, Apple M3 Max, Built-In
Display: HDIT27QD, 2560 x 1440 (QHD/WQHD - Wide Quad High Definition), Main, MirrorOff, Online
Memory Module: LPDDR5, Hynix
AirPort: spairport_wireless_card_type_wifi (0x14E4, 0x4388), wl0: Oct 31 2024 06:18:43 version 23.10.900.20.41.51.176 FWID 01-24228154
IO80211_driverkit-1345.8 "IO80211_driverkit-1345.8" Nov  9 2024 17:02:32
AirPort: 
Bluetooth: Version (null), 0 services, 0 devices, 0 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
USB Device: USB31Bus
USB Device: Anker USB-C Hub
USB Device: Anker USB-C Hub
USB Device: Anker USB-C Hub
USB Device: USB Storage
USB Device: Anker USB-C Hub
USB Device: Anker USB-C Hub
USB Device: Anker USB-C Hub
USB Device: USB31Bus
USB Device: USB31Bus
Thunderbolt Bus: MacBook Pro, Apple Inc.
Thunderbolt Bus: MacBook Pro, Apple Inc.
Thunderbolt Bus: MacBook Pro, Apple Inc.

FedericoRedaelli, 2024-12-28:
Same issue!

### Version of PrusaSlicer
2.8.1 and 2.9.0

### Operating system
macOS 15.2

### CPU
M2

### Printer model
MK3, MK4, MINI

### Notice:
I use the 2.8.0 version on a disk image. This version works fine.

Jan-Soustruznik, 2025-01-02:
Hello, @dustjackettech ,  @FedericoRedaelli , could you share your project file(3MF), please? Thank you

dustjackettech, 2025-01-03:
Since .3MF files aren't one of the "supported" file types for upload, what do you suggest?

Jan-Soustruznik, 2025-01-06:
@dustjackettech Please try renaming the "3MF" to "ZIP" and then upload it. Thank you!

dustjackettech, 2025-01-06:
Or you could just click the link to the hosted location of the file on Printables.

tmicke, 2025-01-13:
Same issue here.
M1 Mac, macOS 15.2
PS 2.9.0
XL 5T

kurtgluck, 2025-05-12:
[benchy5col.4mmfast-440percent-2025v3.3mf.zip](https://github.com/user-attachments/files/20175425/benchy5col.4mmfast-440percent-2025v3.3mf.zip)

When I try to add the text "MADELEINE" to the other side of the bow of this 440% benchy, I add the text - change the letters to "MADELEINE" try to position it to match, edit text window I say USE SURFACE and PER GLIPH - it dies.   Does this often.  There is some problem in the the text positioning on curved surfaces.  Most likely there is a divide by zero or similar uncaught exception.

kurtgluck, 2026-09-18:
This issue should not be closed until 3 weeks after the actual release of 3,0 not an Alpha or Beta release.  This has been a persistent issue with Prusaslicer.