prusa3d/PrusaSlicer #13818: M109 ignored

hugomsmarreiros, 2024-12-21:
### Description of the bug

i have a slight modified start gcode for my 2.5S one of the thins is an m109 after mesh level, with slicer 2.9 M109 is ignored if placed after the G80

### Project file & How to reproduce

[WC.zip](https://github.com/user-attachments/files/18217800/WC.zip)

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.9.0

### Operating system

W11

### Printer model

Prusa 2.5S

gaiuscosades, 2024-12-22:
I have no special knowledge about the issue, but is'nt it most likely a firmware issue of the 2.5S rather than a problem with PrusaSlicer?
Do you have any reasons to think that somehow the slicer is especifically misbehaving in this case?

hugomsmarreiros, 2024-12-23:
Hi, no problem with FW, just use the project I've uploaded, the start gcode is the same I've been using for a long time and has always worked, generate a gcode and open it and you'll see the m109 is missing, if you place it before the mesh level it will be ok but that's not how I want it.

closed 2024-12-25 by hugomsmarreiros