prusa3d/PrusaSlicer #13820: Can not access directories which are symbolic links.

dhpinkerton, 2024-12-21:
### Description of the bug

When importing a STL file I can no longer access symlink directories.

File -> Import -> Import STL

My Documents directory is no longer visible.

As per attached screen shots.  

I also can no longer Export G-code to a symlink directory.

No longer able to open an STL file by double clicking the file.  No longer any Desktop Intergration?

### Project file & How to reproduce

[PrusaSlicer-BugReport_DHP.zip](https://github.com/user-attachments/files/18218713/PrusaSlicer-BugReport_DHP.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.9.0

### Operating system

Fedora 41

### Printer model

Creality Ender 5 Pro

theopensourcerer, 2024-12-21:
+1. Affecting me too. Just installed 2.9.0 using flatpak and I can't access my "projects" directory which is a symlink to another physical drive on my PC. Am running Ubuntu 20.04.

jantastique, 2024-12-21:
+1 same here. 2.9.0 flatpak Mint 22. 
NFS shares are not available for import. Receiving "Loading of model file failed" by drag & drop from NFS. 
Basically it seems I'm only able to access files in my home directory.

jantastique, 2024-12-22:
> +1 same here. 2.9.0 flatpak Mint 22. NFS shares are not available for import. Receiving "Loading of model file failed" by drag & drop from NFS. Basically it seems I'm only able to access files in my home directory.

ok, I figured out the behavior in my case appears due to the sandbox restictions of flatpak. So I needed to add  `--filesystem=/mnt/path/to/workingdir` to the command line starting the app (see [Flatpak permissions](https://docs.flatpak.org/de/latest/sandbox-permissions.html#external-drive-access)).

I successfully tested this with symlinks as well, by adding their paths as described.

Since I have no experience with flatpak (actually avoiding it whenever possible), I can't tell if this is a good solution, but works for me in this case.

dhpinkerton, 2024-12-22:
I've managed to fix mine by...

flatpak override --user --filesystem=host com.prusa3d.PrusaSlicer

To give the app access to all filesystems.

SachCZ, 2024-12-27:
Hello, as you correctly determined, this is a flatpak permissions issue. Personally I would recommend [flatseal](https://flathub.org/apps/com.github.tchx84.Flatseal) to add paths you want permanently.

[image]

vxrexx, 2024-12-29:
I have a similar error: network drives (which are connected to Gigolo) can no longer be used, they are no longer displayed.
(Debian 12.8)

andrewmeyer, 2025-01-03:
can confirm, this is a permissions issue I was able to fix via flatseal, thank you @SachCZ

dbyy, 2025-01-06:
> I've managed to fix mine by...
> 
> flatpak override --user --filesystem=host com.prusa3d.PrusaSlicer
> 
> To give the app access to all filesystems.

I think there is another problem with the new version of Prusaslicer (v2.9.0).
1. the problem which lies actually with flatpak = access permission to directory on a NAS drive.
2. the problem with the pre-selection of file extension in the "import file" dialog (CTRL-I). I'm using a symlink on my home drive which is pointing to the directory on a NAS drive where all my models are located. Since the pre-selected file extension in this dialog (the DDL) makes it impossible to activate this link (to the NAS). The DDL should have an option "all file types" which would make the symlink visible. Then, with the required permissions set in flatpak, I could easily access all my models.
Does that makes sense? 
Would that be a feature request or a reported bug?

SachCZ, 2025-01-07:
Hello @dbyy,
Addressing the second problem:
Was it ever the case that the link was visible? If so then it is a bug. Otherwise it is a feature request. In any case please create a new issue. As this one is fixed by fixing permissions for the OP.

closed 2026-09-29 by prusaslicer-bot[bot]