prusa3d/PrusaSlicer #13823: VARIABLE LAYER HEIGHT CRASHING

JERREDV, 2024-12-22:
### Description of the bug

VARIABLE LAYER HEIGHT CRASHING UPON CLICKING THE FUNCTION BUTTON.

ONCE I TRY AND USE THE VARIABLE LAYER HEIGHT FUNCTION THE APP CRASHES AND CLOSES.

I CHECKED THE FORUMS BUT I COULD NOT FIND A WAY TO RESOLVE THIS ISSUE WITH MY WINDOWS SURFACE PRO 11 COMPUTER.

THE APP WILL CRASH WITH ANY PROJECT I TRY ASWELL AS OLDER VERSIONS OF PRUSA SLICER.

### Project file & How to reproduce

[DRAWERS (FINAL) READY TO PRINT.zip](https://github.com/user-attachments/files/18220897/DRAWERS.FINAL.READY.TO.PRINT.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.9.0

### Operating system

WINDOWS 11. PROCESSOR: Snapdragon(R) X 10-core X1P64100 @ 3.40 GHz   3.42 GHz.... SYSTEM TYPE: 64-bit operating system, ARM-based processor

### Printer model

PRUSA MK4S

TomDegreef, 2025-01-03:
Same issue on Lenovo Yoga with Snapdragon CPU and PrusaSlicer 2.9.0

WashUDave, 2025-02-04:
I am getting the same issue. On Surface Pro 11.

diamondhoodgem, 2025-05-05:
i also get this issue on a surface laptop 7 with snapdragon x

closed 2026-09-29 by prusaslicer-bot[bot]