prusa3d/PrusaSlicer #13824: Variable name not recognized

billyd60, 2024-12-22:
### Description of the bug

I use the following line in my startup gcode:

ROME_START_PRINT EXTRUDER_TEMP=[first_layer_temperature] BED_TEMP=[first_layer_bed_temperature] TOOL=[initial_tool] WIPE_TOWER={wipe_tower} WIPE_TOWER_X={wipe_tower_x} WIPE_TOWER_Y={wipe_tower_y} WIPE_TOWER_WIDTH={wipe_tower_width} WIPE_TOWER_ROTATION_ANGLE={wipe_tower_rotation_angle} COOLING_TUBE_RETRACTION={cooling_tube_retraction} COOLING_TUBE_LENGTH={cooling_tube_length} PARKING_POS_RETRACTION={parking_pos_retraction} EXTRA_LOADING_MOVE={extra_loading_move};

It's always worked in the past. However under version 2.9.0 it throws an error, stating invalid variable name. I believe it is (wipe_tower_x). Now for this particular print, I am not using the wipe tower, but that has never caused an error before.

### Project file & How to reproduce

n/a

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.9.0

### Operating system

Windows 10

### Printer model

Vcore3.1 - 500 with ROME

neophyl, 2024-12-22:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/13684

Jan-Soustruznik, 2025-01-02:
closed as duplicate of  https://github.com/prusa3d/PrusaSlicer/issues/13684

closed 2025-01-02 by Jan-Soustruznik