prusa3d/PrusaSlicer #13825: Significantly worse generation of Aligned seams in 2.9.0

Karlosak, 2024-12-22:
### Description of the bug

After upgrading to PrusaSlicer 2.9.0 from 2.8.1 I'm noticing that the "Aligned" seam position is creating a much coarser, non-aligned seams, which leads to poorer, regressed print quality
[image]
[image]

### Project file & How to reproduce

[Clip-7pcs.zip](https://github.com/user-attachments/files/18222681/Clip-7pcs.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.9.0

### Operating system

Windows 10 64bit

### Printer model

Prusa MK4S

Jan-Soustruznik, 2025-01-02:
Hello, @Karlosak, thank you for your report. I’ve created an internal issue (SPE-2634). We’ll try to fix in some of new release. Thank you, have a nice day.

Karlosak, 2025-01-02:
Thank you @Jan-Soustruznik !

dougemon, 2025-10-19:
Hi there, I am having similar issues as shown above in PrusaSlicer 2.9.3.
Hope this gets resolved soon

closed 2026-09-29 by prusaslicer-bot[bot]