prusa3d/PrusaSlicer #13833: + de 7000 lignes de commentaires en version 2.9.0

18pool49, 2024-12-23:
### Description of the bug

quand l'on désire lire le Gcode avec "Notepad++" ou "Bloc-notes" l'on trouve 7359 ligne de commentaires en langage étrange ???

[image]

en version 2.9.0 ?????     en 2.8 je n'avais pas remarqué cela 

### Project file & How to reproduce

[XL 4  bree-fullmodel-2.9 4 couleur pas T5 projet.zip](https://github.com/user-attachments/files/18231352/XL.4.bree-fullmodel-2.9.4.couleur.pas.T5.projet.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.9.0

### Operating system

W11

### Printer model

XL4  le T5 est bleu >>> carte défectueuse

neophyl, 2024-12-23:
Those are the text encoding of the thumbnail images.  As such its not a bug.

18pool49, 2024-12-24:
merci pour cette précision ; 
j'ai  trouvé la solution >> en supprimant la définition de la vignette dans le Gcode  
[image]
comme cela que je retrouve une lecture plus pratique du Gcode

closed 2024-12-24 by 18pool49