prusa3d/PrusaSlicer #13834: Multiple dovetails in one cut

henxdl, 2024-12-23:
**Is your feature request related to a problem? Please describe.**
Unfortunately, when printing something like a large torus or hexagon, you cannot cut it with two dovetails because it is only implemented for 1 dovetail per cut. This is very inconvenient when cutting into dovetails, and forces you to use an actual CAD software.
**Describe the solution you'd like**
I would like a "dovetail number" selector that allows you to choose how many dovetails you want in the cut, and then move multiple dovetails to their desired spots.
**Describe how it would work**
This is relatively straightforward, but you would be able to select each dovetail and move them to the correct spot.
**Describe alternatives you've considered**
I have not considered any other alternatives, and I do not think there are any other solutions to achieve this specific result.
**Additional context**
Example screenshot of the issue:

[image]
If you try to make the cut, you are left with only 1 joint connected.

neophyl, 2024-12-23:
Duplicate of multiple issues.  Please check for existing issues before opening new ones.  It just clutters up the repo and doesn't get things done any quicker.  Go add your support to the earliest version.

https://github.com/prusa3d/PrusaSlicer/issues/13544
https://github.com/prusa3d/PrusaSlicer/issues/12655
https://github.com/prusa3d/PrusaSlicer/issues/12077

henxdl, 2024-12-24:
I apologize for that. I actually felt very surprised that "nobody" had made an issue on it yet. Guess I didn't search very thoroughly.

henxdl, 2025-01-02:
Incorrectly closed.

closed 2025-01-02 by henxdl