prusa3d/PrusaSlicer #13835: Segmentation fault (core dumped) when opening Physical Printer tab on Linux Mint 21.3

gknapp, 2024-12-23:
### Description of the bug

Opening PrusaSlicer 2.8.1 (using .AppImage) and selecting the Physical Printer tab initially shows the klipper / mainsail Initialising bar but then crashes the whole application. I ran the .AppImage via the terminal to see what console output I'd see, all I get is "Segmentation fault (core dumped)".

I've a separate issue with 2.9.0 using flatpak where PrusaSlicer will not resolve my mDNS hostname (xyz.local) but 2.8.1 will - so it behaves differently (no crash).

### Project file & How to reproduce

Opening PrusaSlicer 2.8.1, with a printer configured for Host Type 'Klipper (via Moonraker)' and a hostname specified:

- Select the Physical Printer tab

[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.8.1

### Operating system

Linux Mint 21.3

### Printer model

FlyingBear Ghost 6

gknapp, 2024-12-23:
This has happened on earlier versions:

Klipper: v0.12.0-404-g80d185c94
Moonraker: v0.9.3-3-gccfe32f2
Mailsail: v2.13.0

Jan-Soustruznik, 2025-01-03:
Hello, @gknapp, @SorrelArticulata , do you have set the API key/ Password for the connection? Thank you, have a nice day

AlexanderS, 2025-01-11:
This is caused by c3ca39d5c5458a58402abe737ef90c061b76aeb5. I can reproduce the issue in surf and epiphany (two webkit browsers) if I manually set `WEBKIT_DISABLE_COMPOSITING_MODE=1` for these browsers and navigating to a mainsail page.

Just for reference, this is the backtrace for the segfault:

```
#0  webkitWebViewBaseEnterAcceleratedCompositingMode(_WebKitWebViewBase*, WebKit::LayerTreeContext const&) () at ./Source/WebKit/UIProcess/API/gtk/WebKitWebViewBase.cpp:2938
#1  0x00007fffedd74ca8 in WebKit::WebPageProxy::enterAcceleratedCompositingMode(WebKit::LayerTreeContext const&) () at ./Source/WebKit/UIProcess/WebPageProxy.cpp:10743
#2  0x00007fffedeaf4ee in WebKit::DrawingAreaProxyCoordinatedGraphics::enterAcceleratedCompositingMode(WebKit::LayerTreeContext const&) () at ./Source/WebKit/UIProcess/CoordinatedGraphics/DrawingAreaProxyCoordinatedGraphics.cpp:237
#3  0x00007fffeda1a440 in _ZZN3IPC18callMemberFunctionIN6WebKit16DrawingAreaProxyES2_FvmRKNS1_16LayerTreeContextEESt5tupleIJmS3_EEEEvPT_MT0_T1_OT2_ENKUlDpOT_E_clIJmS3_EEEDaSI_ () at ./Source/WebKit/Platform/IPC/HandleMessage.h:135
#4  _ZSt13__invoke_implIvZN3IPC18callMemberFunctionIN6WebKit16DrawingAreaProxyES3_FvmRKNS2_16LayerTreeContextEESt5tupleIJmS4_EEEEvPT_MT0_T1_OT2_EUlDpOT_E_JmS4_EESA_St14__invoke_otherOSC_DpOT1_ () at /usr/include/c++/12/bits/invoke.h:61
#5  _ZSt8__invokeIZN3IPC18callMemberFunctionIN6WebKit16DrawingAreaProxyES3_FvmRKNS2_16LayerTreeContextEESt5tupleIJmS4_EEEEvPT_MT0_T1_OT2_EUlDpOT_E_JmS4_EENSt15__invoke_resultISA_JDpT0_EE4typeEOSA_DpOSM_ () at /usr/include/c++/12/bits/invoke.h:96
#6  _ZSt12__apply_implIZN3IPC18callMemberFunctionIN6WebKit16DrawingAreaProxyES3_FvmRKNS2_16LayerTreeContextEESt5tupleIJmS4_EEEEvPT_MT0_T1_OT2_EUlDpOT_E_S9_JLm0ELm1EEEDcOSA_OSC_St16integer_sequenceImJXspT1_EEE () at /usr/include/c++/12/tuple:1852
#7  _ZSt5applyIZN3IPC18callMemberFunctionIN6WebKit16DrawingAreaProxyES3_FvmRKNS2_16LayerTreeContextEESt5tupleIJmS4_EEEEvPT_MT0_T1_OT2_EUlDpOT_E_S9_EDcOSA_OSC_ () at /usr/include/c++/12/tuple:1863
#8  IPC::callMemberFunction<WebKit::DrawingAreaProxy, WebKit::DrawingAreaProxy, void (unsigned long, WebKit::LayerTreeContext const&), std::tuple<unsigned long, WebKit::LayerTreeContext> >(WebKit::DrawingAreaProxy*, void (WebKit::DrawingAreaProxy::*)(unsigned long, WebKit::LayerTreeContext const&), std::tuple<unsigned long, WebKit::LayerTreeContext>&&) () at ./Source/WebKit/Platform/IPC/HandleMessage.h:133
#9  IPC::handleMessage<Messages::DrawingAreaProxy::EnterAcceleratedCompositingMode, WebKit::DrawingAreaProxy, WebKit::DrawingAreaProxy, void (unsigned long, WebKit::LayerTreeContext const&)>(IPC::Connection&, IPC::Decoder&, WebKit::DrawingAreaProxy*, void (WebKit::DrawingAreaProxy::*)(unsigned long, WebKit::LayerTreeContext const&)) () at ./Source/WebKit/Platform/IPC/HandleMessage.h:235
#10 WebKit::DrawingAreaProxy::didReceiveMessage(IPC::Connection&, IPC::Decoder&) () at DerivedSources/WebKit/DrawingAreaProxyMessageReceiver.cpp:48
#11 0x00007fffedce9c2a in IPC::MessageReceiverMap::dispatchMessage(IPC::Connection&, IPC::Decoder&) () at ./Source/WebKit/Platform/IPC/MessageReceiverMap.cpp:129
#12 0x00007fffeddf24aa in WebKit::WebProcessProxy::didReceiveMessage(IPC::Connection&, IPC::Decoder&) () at ./Source/WebKit/UIProcess/WebProcessProxy.cpp:1212
#13 0x00007fffedce4785 in IPC::Connection::dispatchMessage(WTF::UniqueRef<IPC::Decoder>) () at ./Source/WebKit/Platform/IPC/Connection.cpp:1451
#14 0x00007fffedce512c in IPC::Connection::dispatchMessage(WTF::UniqueRef<IPC::Decoder>) () at ./Source/WebKit/Platform/IPC/Connection.cpp:1408
#15 IPC::Connection::dispatchIncomingMessages() () at ./Source/WebKit/Platform/IPC/Connection.cpp:1563
#16 0x00007fffebf15555 in WTF::Function<void ()>::operator()() const () at ./Source/WTF/wtf/Function.h:82
#17 WTF::RunLoop::performWork() () at ./Source/WTF/wtf/RunLoop.cpp:147
#18 0x00007fffebfaa899 in operator() () at ./Source/WTF/wtf/glib/RunLoopGLib.cpp:80
#19 _FUN() () at ./Source/WTF/wtf/glib/RunLoopGLib.cpp:82
#20 0x00007fffebfab55f in operator() () at ./Source/WTF/wtf/glib/RunLoopGLib.cpp:53
#21 _FUN() () at ./Source/WTF/wtf/glib/RunLoopGLib.cpp:56
#22 0x00007ffff211c67f in g_main_dispatch (context=0x555557bb8630) at ../../../glib/gmain.c:3454
#23 g_main_context_dispatch (context=context@entry=0x555557bb8630) at ../../../glib/gmain.c:4172
#24 0x00007ffff211ca38 in g_main_context_iterate (context=0x555557bb8630, block=block@entry=1, dispatch=dispatch@entry=1, self=<optimized out>) at ../../../glib/gmain.c:4248
#25 0x00007ffff211ccef in g_main_loop_run (loop=loop@entry=0x55556a2653e0) at ../../../glib/gmain.c:4448
#26 0x00007ffff2408495 in gtk_main () at ../../../gtk/gtkmain.c:1329
#27 0x00007ffff356df85 in wxGUIEventLoop::DoRun() (this=0x55556a2abac0) at ./src/gtk/evtloop.cpp:61
#28 0x00007ffff3cd7fad in wxEventLoopBase::Run() (this=0x55556a2abac0) at ./src/common/evtloopcmn.cpp:87
#29 0x00007ffff3ca2a9b in wxAppConsoleBase::MainLoop() (this=0x555557b9f650) at ./src/common/appbase.cpp:381
#30 0x00007ffff3d1fec7 in wxEntry(int&, wchar_t**) (argc=<optimized out>, argv=<optimized out>) at ./src/common/init.cpp:508
#31 0x0000555556847b1e in Slic3r::GUI::GUI_Run(Slic3r::GUI::GUI_InitParams&) ()
#32 0x0000555555dce080 in Slic3r::CLI::run(int, char**) ()
#33 0x0000555555d84d2c in main ()
```

jiskim, 2025-04-24:
i had a simillar issue with bambulab/orcaslicer and this comment fixed my issue; and this fixed my issue with prusaslicer too
 ( tested on 2.9.0 )

https://github.com/bambulab/BambuStudio/issues/4626#issuecomment-2676350419

> For:
> 
> * Bambu Studio 1.10.1 flatpak from flathub
> * Fedora 41
> * Nvidia GTX 1080ti with Nvidia driver 565.77
> 
> These env vars work for me applied with flatseal:
> 
> * WEBKIT_DISABLE_DMABUF_RENDERER=1
> * GALLIUM_DRIVER=zink
> * MESA_LOADER_DRIVER_OVERRIDE=zink
> * __GLX_VENDOR_LIBRARY_NAME=mesa
> * __EGL_VENDOR_LIBRARY_FILENAMES=/usr/lib/x86_64-linux-gnu/GL/default/share/glvnd/egl_vendor.d/50_mesa.json
> 
> The preview window clearly works will hardware acceleration. No choppiness.
> 
> Hope Bambu Lab can resolve this issue soon for everyone - without overrides.

if you are using the flatpack version to add the env vars use flatseal.

here is what i did : 
 1. get that : https://flathub.org/apps/com.github.tchx84.Flatseal
 2. open the prusaslicer tab ( in my case i installed it in the system so it show automatically ) 
 3. scroll down to the Environment section and add the env vars listed earlier 
 4. close falt seal and start prusaslicer and it should now work.

just leaving that here since that bug was annoying me to no end for the last few months

PimpMyPizza, 2025-06-10:
Thanks a lot!
```bash
flatpak override --user com.bambulab.BambuStudio \
  --env=WEBKIT_DISABLE_DMABUF_RENDERER=1 \
  --env=GALLIUM_DRIVER=zink \
  --env=MESA_LOADER_DRIVER_OVERRIDE=zink \
  --env=__GLX_VENDOR_LIBRARY_NAME=mesa \
  --env=__EGL_VENDOR_LIBRARY_FILENAMES=/usr/lib/x86_64-linux-gnu/GL/default/share/glvnd/egl_vendor.d/50_mesa.json
```
did it for me too.
- Fedora 42 KDE
- RTX3060
- Flatpak BambuStudio 2.1.0

closed 2026-09-29 by prusaslicer-bot[bot]