prusa3d/PrusaSlicer #13836: Flow::spacing() produced negative spacing. Did you set some extrusion width too small? in version 2.9

balloonhotair, 2024-12-23:
### Description of the bug

see this forum entry for all info.
https://forum.prusa3d.com/forum/prusaslicer/slicer-2-9-bug/#post-730579

### Project file & How to reproduce

see this forum entry for all info.
https://forum.prusa3d.com/forum/prusaslicer/slicer-2-9-bug/#post-730579

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.9

### Operating system

windows 7

### Printer model

xl

Jan-Soustruznik, 2025-01-02:
Hello, @balloonhotair, thank you for your report. It seams that is similar issue of [lh_negative_extrusion_width](https://dev.prusa3d.com/issues/?jql=gitBranch+in+(lh_negative_extrusion_width)) .

Jan-Soustruznik, 2025-01-10:
Closed as duplicate of https://github.com/prusa3d/PrusaSlicer/issues/13429