prusa3d/PrusaSlicer #13838: prusa connect connection issues

ac2112, 2024-12-24:
### Description of the bug

When the computer goes to sleep, if you have Prusa Slicer open, it disconnects from Prusa Connect. This means you have to go through the process of re-entering the password and 2-factor authentication.

Please, Please, Please, add a feature to remember the password and the ability to trust my computer so I don't have to constantly go through the authentication process.  Just like a standard website.

I'm at the point where I need to turn off 2-factor authentication.

### Project file & How to reproduce

Just let your computer go to sleep.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.8.1+win64

### Operating system

Windows 11

### Printer model

XL

halofx, 2024-12-30:
YES PLEASE!! My computers don't go into full sleep mode, but are constantly getting logged out of PrusaConnect. Maybe it has to do with me using PrusaConnect from multiple machines? But it is why I refuse to turn on 2-Factor for this account. I already have a secure password, but it is in BitWarden. So I have to daily go and copy out the password to log in.

Jan-Soustruznik, 2025-02-19:
Hello, @halofx @ac2112 , Should be fixed in [PrusaSlicer 2.9.1-alpha1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.9.1-alpha1). Closing.

closed 2025-02-19 by Jan-Soustruznik