prusa3d/PrusaSlicer #13839: Can't log in, only on "printables.com".

Dre91Dre91, 2024-12-24:
### Description of the bug

can't log in  anywhere except printables.com.
wont even allow me to  type log-in credentials and only shows "failed log-in"

### Project file & How to reproduce

[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

latest 2.90

### Operating system

win 10

### Printer model

i3 mk3s +

halofx, 2024-12-30:
What happens when you click the Log In in the top right? It is very annoying that PrusaSlicer keeps losing the log in credentials, and requiring them to be re entered, almost daily for me.

kocikdav, 2025-01-02:
Hello @Dre91Dre91. Thank you for reporting this issue. 
First, please explain what  "can't log in anywhere except printables.com" means. 
Second, if you are seeing "failed login" notification right after clicking on "Log in" button on Windows, please try following:
- Close PrusaSlicer. 
- Locate a folder _C:\Users\User\AppData\Local\PrusaSlicer_. Inside you should see only folder _EBWebView_. 
- Delete _EBWebView_ folder.

kocikdav, 2025-01-03:
Hello @halofx I see you have commented under several issues while I suspect your issue is similar to https://github.com/prusa3d/PrusaSlicer/issues/13883. If it really seems to be your issue, lets continue there, if not lets start separate issue.

closed 2026-09-29 by prusaslicer-bot[bot]