prusa3d/PrusaSlicer #13844: Can not import files

DenEric, 2024-12-26:
### Description of the bug

With the version 2.9.0 of the prusaslicer I can not access all drives on my machine to load files. Even linked files in the home directory are not accessible.

### Project file & How to reproduce

'File' -> import 
Directory is not found

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.9.0

### Operating system

Linux

### Printer model

MK4

SachCZ, 2024-12-27:
Hello, this is intended behavior of flatpak. It limits the application access to the filesystem. You can change this using [flatsteal](https://flathub.org/apps/com.github.tchx84.Flatseal). You can add any desired paths to `Other files` in the following section:
[image]

Jan-Soustruznik, 2025-01-02:
closed as duplicate of https://github.com/prusa3d/PrusaSlicer/issues/13820

closed 2025-01-02 by Jan-Soustruznik