prusa3d/PrusaSlicer #13846: Disable / switch off "printables" tab from main interface.

0subzero, 2024-12-26:
[image]

Hi all.
I don't have internet on my PC.
So it's quite annoying new tab.
Any way to disable it?
Thanks.

neophyl, 2024-12-26:
No there isn't. As yet.  Your issues is a duplicate of https://github.com/prusa3d/PrusaSlicer/issues/13635 btw.  Please do search for existing issues before opening new ones.

0subzero, 2024-12-28:
sorry. i did search "Printables" but didn't find this post.
okay. you can delete it ... or close ...

closed 2025-01-06 by TerPuc

kubispe1, 2025-01-06:
Regarding this post, I would like to provide a brief explanation of our goals for the new Printables tab:
If you're interested: https://github.com/prusa3d/PrusaSlicer/issues/13635#issuecomment-2573452456