prusa3d/PrusaSlicer #13847: drag/drop stl

jopen-ux, 2024-12-26:
### Description of the bug

error popup : loading of a model file failed
[image]

### Project file & How to reproduce

Select a file in exploer (kde : dolphin) , drop the file in prisaa slicer bed area.
Was working with all older versions
KO in 2.9.0 all times

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.9.0

### Operating system

Kubunt24.04

### Printer model

all (not printer dependent

Jan-Soustruznik, 2025-01-02:
Hello, @jopen-ux, could you share your model file that evoke error message, please?

JCruk, 2025-02-02:
I have this issue as well.

I see the same error with drag and drop. I can open the same file if I use import - however, many directories aren't visible with the import dialogue. Only a couple of directories, that are within my home directory, are visible.

I'm on Ubuntu 24.04.1
Installed the latest flatpak
Uninstalled and re-installed
Issue remains.

This is the .stl file I am using:
https://www.thingiverse.com/thing:1680619

neophyl, 2025-02-02:
Could it be the flatpack permissions issue ?  see this stickied issue from the top of the forum https://github.com/prusa3d/PrusaSlicer/issues/13820  
The description of not visible directories sounds like it.

jopen-ux, 2025-02-02:
Yes il is Use flatseal to give permissions to over filesystems monted permanently. Flatpack is like snap's a nightmare... Apt install was better... ...

jopen-ux, 2025-02-02:
Flatpack problem. Use flatseal to give permissions to other filesystems. Do not know why PRUSA has limited permissions...On m'y own expérience with flatpack this was the first Time I had to use flatseal. And as I am (still) using kubuntu and have removed all thies bullshit snaps, I have many flatpack in /opt

https://www.thingiverse.com/thing:1680619

closed 2025-02-05 by SachCZ