prusa3d/PrusaSlicer #13848: Linux flatpak uses different DNS config than host and doesn't report effectively when broken

enygren, 2024-12-26:
### Description of the bug

The Linux flatpak packaging of PrusaSlicer 2.9.0 uses an alternate /etc/nsswitch.conf from that of the host.  This forces the use of systemd-resolved if it is running, even if its usage isn't configured on the host via /etc/nsswitch.conf.  The result is that if systemd-resolved is running but misconfigured on the host but worked around via /etc/nsswitch.conf, DNS will break within the flatpak container causing anything needing network access to not work effectively.

Note that while this is not strictly a bug in the container packaging, it took me hours of debugging to trace this down since DNS was working fine on the host (due to nsswitch.conf bypassing systemd-resolved and using resolv.conf) but breaking within the container (due to using its own nsswitch.conf).

### Project file & How to reproduce

If systemd-resolved is in a bad state, DNS in the flatpak container won't work.  The user won't get any indication of this other than broken networking.

Some ways to test:

`flatpak run --command=curl com.prusa3d.PrusaSlicer -v 'https://www.prusa3d.com/'`

(should work, but will return a DNS error if broken)

To see if DNS is working on the host with systemd-resolved, if this fails on the host then this needs to get fixed by the user before the container will work:

`resolvectl query example.com`

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.9.0

### Operating system

Ubuntu 22.04

### Printer model

mk3.5

bdwheele, 2024-12-28:
Probably related to this -- mDNS queries also don't work in the flatpak, so addresses like `prusa-mk4.local` do not resolve.

System nsswitch.conf:
```
hosts:      files myhostname mdns4_minimal [NOTFOUND=return] resolve [!UNAVAIL=return] dns
```
Flatpak nsswitch.conf:
```
hosts: files myhostname resolve [!UNAVAIL=return] dns
```

xiphmont, 2025-01-02:
That's essential but not enough-- the shipped flatpak does not include any support for mDNS, so you're configuring nsswitch to look for support that's not there.  It would need an extension module that allows access to the host resolver/avahi.  I think those do exist.

xiphmont, 2025-01-03:
> I looked into this after updating and hitting the issue myself, and as best as i could determine it is impossible to have additional resolvers in flatpak apps right now, because theres no extensions capable of adding the mdns nss modules to `/usr/lib` and possibly modifying nsswitch.conf too?

Would the flatpak "network-access-mDNS" extension module work for our needs?  It's a little hectic here right now, but I'll give it a try if you don't beat me to it.

xiphmont, 2025-01-03:
Oh no... I have been trolled by Google's new AI search summary providing instructions, example code (and links!) to a flatpak extension--- that simply does not exist.  I am a little flabbergasted.

xiphmont, 2025-01-04:
Indeed, properly embarrassed and surprised it got me so easily with a hallucination like that.

In any case, the ground truth appears to be 'it's tricky talking to the host Avahi, or any of the other things that talk to Avahi, because security' and that has led to no one tackling it.  Apparently importing/shipping flatpak's own mDNS would conflict with the host's given other aspects of network sharing.  Looks like that will not get solved soon unless someone's looking for a really unfulfilling new hobby.  Writing a portal isn't so hard.  Security is. :-(

vkorenev, 2026-09-19:
This issue is not fixed in PrusaSlicer 3.0.0 alpha. I installed a flatpak from flathub-beta and mDNS is still not working.