prusa3d/PrusaSlicer #13849: Linux File explorer when adding file (CTRL+I)

jopen-ux, 2024-12-26:
### Description of the bug

The prusasilcer internal explorer window do not showing
[pb2.9.0.pdf](https://github.com/user-attachments/files/18253948/pb2.9.0.pdf)
 any more the filesystems mounted under root (/) and not showing symlinks of these partitions.

### Project file & How to reproduce

click on add button go back to /

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.9.0

### Operating system

kubuntu 24.04

### Printer model

all (XL, MK4S, MK3, volumic

SachCZ, 2024-12-27:
Hello this is a security feature of flatpak. It limits what the application can access. I recommend you use flatsteal to change the permissions. See https://github.com/prusa3d/PrusaSlicer/issues/13844#issuecomment-2563584541.

Jan-Soustruznik, 2025-01-02:
closed as duplicate of https://github.com/prusa3d/PrusaSlicer/issues/13820

closed 2025-01-02 by Jan-Soustruznik