prusa3d/PrusaSlicer #13852: Need a way to select MMU print order, especially first layer.

mj8w, 2024-12-26:
Print order is important for small details on the first layer, especially for small signs and miniatures, mainly because the first color printed has a bit of elephants foot, and "grabs" the real-estate on the build plate. Further colors smoosh around the first color.

Please add a "first layer print order" to Print Settings | multi-extruder page, i.e. a text entry for comma separated values, defaults to "1,2,3,4,5", or similar, 
- tooltip would specify this applies to first layer only
- if the text in the entry is invalid, render the text in red, and default to 1,2,3,4,5 internally.

- the only operational difference is to move the colors around in the Gcode (and of course the MMU instructions)

Of course the user can reload all the filaments, but this is a hassle. It takes a few minutes for each roll of filament.

Thanks for considering this change.

halofx, 2024-12-30:
I am not saying I wouldn't use this feature, but the current operation order when more than one color is on the first plate is to go in order of the extruders.  I agree that previously it was a hassle to have to physically move the filaments to get the order needed. The recent addition of the Filament Mappings before print start has mitigated this for me on the MK4/MMU and XL. Granted, that to my knowledge this only applies to the Prusa machines with the xLCD (MK4/XL), I am not sure how this works on the MK3/MMU? And would still be an issue on non-Prusa machines. @mj8w - Are you able to use the Filament Mapping Menu on the machine?

jalicz, 2025-01-08:
That's a very good idea. I also have a problem with the correct order of the extruder when starting to print. I have to manually change the color in the first extruder to start printing with the needed color to eliminate unnecessary color changes and improve the print quality of the first layer. I need to print the surrounding areas and large parts first and then fill in the gaps of the smaller parts. However, if there is a color in the first extruder for these secondary parts, I need to change the colors first, I haven't found a way to change this, only by manually adjusting the GCODE

kkeennpp, 2025-02-17:
Please implement this! I really like my MMU and printing in multicolor, but having to change the filament order each time I want to change the color print order for the first layer is very annoying epically without a prefeeder like the Bambu has. The first layer color order greatly affects how clean edges can be in fine detail since the filament slightly squishes out while printing. If I print the small details first then do the rest of the fill, it has a much crisper line and more uniform print.

MudiStar, 2025-02-20:
Same problem here
I would really appreciate this feature

sebbl2, 2025-03-05:
Especially small signs would benefit if this works. Start first layer with color 3 (red ) and afterwards everything else color 1 (black). The letters mustn't  be squished into the perimeters of color 1.   
I am missing this feature since I bought the mmu3 because its still a lot of work to change the Filament rolls on my cupboard, which is used as an enclosure.
I really appreciate your work and love to see this in future.

Peaches491, 2025-03-20:
Yes, I'd love this feature as well. Particularly for manual filament changes (since I use a single extruder home-built Mendel 90)
Right now,the slicer _insists_ on printing the red (extruder 1) first, and the blue text (extruder 2) second. However there's a second layer that's all red, forcing me to have two filament changes instead of one. 

If I could tell it to print the blue first it would only take one filament change

[image]

steelydev, 2025-05-07:
This is a necessary improvement.  Physically swapping the filaments is a pain when printing small text or features on first layer with MMU3

BurnData, 2025-08-15:
PLEASE implement this change. With the 5 head XL this is a really needed feature. With such an expensive and advanced machine, needing to manually rearrange filaments is absurd. And with the implementation of being able to mix nozzle sizes, I find I have to physically move a print head or a nozzle, which then requires a recalibration, which takes a LOT of time. 

I often use a 0.2 nozzle to imbed fine details in the surface layers of a print which otherwise uses a 0.4 nozzle. Constantly moving the print head and recalibrating is just not a viable option.

ScubaSteveDammU, 2025-09-09:
Yes, please implement this feature.

TomBrodsky, 2026-09-13:
Still interested.

Thann, 2026-10-02:
This feature is essential for virtual extruders and doesnt appear to be fixed in 3.0.0!