prusa3d/PrusaSlicer #13853: Disabling Z lift does not stop Z lifting

BixlerVDL, 2024-12-26:
### Description of the bug

In PrusaSlicer I have set the "Lift height" to 0mm, "ramping lift" is disabled.

During printing, the printer still lifts the Z height during travel moves.

### Project file & How to reproduce

[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.9.0

### Operating system

Windows 10

### Printer model

Prusa XL

neophyl, 2024-12-26:
Fellow user here but you haven't included your project file as requested when opening a new issue so we do not have enough information to give a reasonable answer.

For example what filament profile are you using ?  Filament profiles have an over ride section.  If the filament has a lift defined in the over rides then changing the setting on the extruder is going to do nothing at all.  If we had your project it could have been opened and we could check if that's the case.  There are so many settings a project is pretty much required to debug anything.

BixlerVDL, 2025-01-06:
Thank you, it was indeed a setting in the filament overwrite section.
I never touched that section, it seems strange that it was already set at a certain value in the overwrite section.
Thanks again, printer is working a lot better with little to no stringing with the z lift feature disabled.

closed 2025-01-06 by BixlerVDL