prusa3d/PrusaSlicer #13858: Different material per plate

ScreennameTaken, 2024-12-27:
I made a project in prusa slicer that requires different kinds of materials.

For some reason, i thought that if i placed my objects in different plates, i would be able to have different settings for each plate.
For sure i thought that i would be able to assign different materials.

I then remembered that i read how 2.9 does its own plates, it is just assigning plates by measuring how far the objects are from the area of the first plate.

Since 3mf is basically a zip file with an xml file describing the contents, would it be possible to use new tags that prusa slicer could read (and any slicer that doesn't know what the tags are, ignore) and assign different settings per plate?

halofx, 2024-12-30:
I would love each plate has its own set of defaults for Print Settings, and Filament. I imagine it as adding another level to the Object List interface. Each plate listed, then objects nested under the plate, Then default filament and settings could be applied per plate. Just as the active plate is highlighted in the Model Preview window, and others are greyed, extend that to the Object List. Multiple plates haven't proven that useful to me, without being able to have this extra level of control.
[image]

Tupson444, 2024-12-31:
I find multiple plates most useful when I print many different parts, but with the same settings. This way, if I tweak something, I only need to do it once and then reslice everything (instead of manually repeating the process for separate .3mf files). If I do need different settings, I will make separate files.

But this is just my opinion (for my workflow), and I'm not against the idea.

lukasmatena, 2026-09-01:
Implemented in [3.0.0-alpha11](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_3.0.0-alpha11). Closing.

closed 2026-09-01 by lukasmatena