prusa3d/PrusaSlicer #13859: /usr/include/c++/14.2.1/bits/stl_vector.h:1213: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::front() [with _Tp = unsigned char; _Alloc = std::allocator<unsigned char>; reference = unsigned char&]: Assertion '!this->empty()' failed.

edwloef, 2024-12-28:
### Description of the bug

`/usr/include/c++/14.2.1/bits/stl_vector.h:1213: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::front() [with _Tp = unsigned char; _Alloc = std::allocator<unsigned char>; reference = unsigned char&]: Assertion '!this->empty()' failed.`

Can be consistently reproduced by trying to slice the part `scaled_Left_part_reworked.STL` from https://www.thingiverse.com/thing:5362460 on the built-in `0.10mm DETAIL` setting

[image]

### Project file & How to reproduce

[project.zip](https://github.com/user-attachments/files/18265479/project.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.9.0

### Operating system

Arch Linux

### Printer model

Prusa i3 MK3

ilyxa, 2024-12-28:
```
[ilyxa@wombatneo gato-flexi-print-in-place-stl]$ uname -a 
Linux wombatneo 6.12.7-arch1-1 #1 SMP PREEMPT_DYNAMIC Fri, 27 Dec 2024 14:24:37 +0000 x86_64 GNU/Linux
[ilyxa@wombatneo gato-flexi-print-in-place-stl]$ trizen -Q prusa-slicer 
prusa-slicer 2.9.0-1
[ilyxa@wombatneo gato-flexi-print-in-place-stl]$ prusa-slicer --loglevel 255 --slice 1.stl
[2024-12-28 19:42:55.786374] [0x00007223fdacd080] [debug]   TriangleMesh::repair() started
[2024-12-28 19:42:56.084082] [0x00007223fdacd080] [info]    edge 1 of facet 526000 doesn't match edge 2 of facet 528136
facet (526000)/ first facet
  vertex 0/526000  7.34441757E+01  8.67241211E+01  3.82802658E+01
  vertex 1/526000  7.34514771E+01  8.67195053E+01  3.82832947E+01
  vertex 2/526000  7.33866653E+01  8.67636414E+01  3.82691040E+01
facet (528136)/ second facet
  vertex 0/528136  7.34441757E+01  8.67241211E+01  3.82802658E+01
  vertex 1/528136  7.32572556E+01  8.67386017E+01  3.84865761E+01
  vertex 2/528136  7.33866653E+01  8.67636414E+01  3.82691040E+01
[2024-12-28 19:42:56.084210] [0x00007223fdacd080] [info]    edge 2 of facet 528136 doesn't match edge 1 of facet 526000
facet (528136)/ first facet
  vertex 0/528136  7.34441757E+01  8.67241211E+01  3.82802658E+01
  vertex 1/528136  7.32572556E+01  8.67386017E+01  3.84865761E+01
  vertex 2/528136  7.33866653E+01  8.67636414E+01  3.82691040E+01
facet (526000)/ second facet
  vertex 0/526000  7.34441757E+01  8.67241211E+01  3.82802658E+01
  vertex 1/526000  7.34514771E+01  8.67195053E+01  3.82832947E+01
  vertex 2/526000  7.33866653E+01  8.67636414E+01  3.82691040E+01
[2024-12-28 19:42:56.248245] [0x00007223fdacd080] [debug]   TriangleMesh::repair() finished
[2024-12-28 19:42:56.409957] [0x00007223fdacd080] [info]    Starting the slicing process. Resident memory: 94MB; Shared memory: 37MB; Private memory: 57MB; Peak memory usage: 304MB
10 => Processing triangulated mesh
[2024-12-28 19:42:56.410027] [0x00007223fdacd080] [info]    Slicing volumes... Resident memory: 94MB; Shared memory: 37MB; Private memory: 57MB; Peak memory usage: 304MB
[2024-12-28 19:42:56.413181] [0x00007223fdacd080] [debug]   slice_mesh to polygons
/usr/include/c++/14.2.1/bits/stl_vector.h:1213: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::front() [with _Tp = unsigned char; _Alloc = std::allocator<unsigned char>; reference = unsigned char&]: Assertion '!this->empty()' failed.
Aborted (core dumped)
[ilyxa@wombatneo ~]$ coredumpctl info 29899
           PID: 29899 (slic3r_main)
           UID: 1000 (ilyxa)
           GID: 1000 (ilyxa)
        Signal: 6 (ABRT)
     Timestamp: Sat 2024-12-28 19:42:56 MSK (3min 37s ago)
  Command Line: prusa-slicer --loglevel 255 --slice 1.stl
    Executable: /usr/bin/prusa-slicer
 Control Group: /user.slice/user-1000.slice/session-1.scope
          Unit: session-1.scope
         Slice: user-1000.slice
       Session: 1
     Owner UID: 1000 (ilyxa)
       Boot ID: 333ae6a588af4b0c9583e35fb4862676
    Machine ID: 651f49ec3f874297b4c4f3472cdc661b
      Hostname: wombatneo
       Storage: /var/lib/systemd/coredump/core.slic3r_main.1000.333ae6a588af4b0c9583e35fb4862676.29899.1735404176000000.zst (present)
  Size on Disk: 65.2M
       Message: Process 29899 (slic3r_main) of user 1000 dumped core.
```
Confirmed on another model.

ilyxa, 2024-12-28:
distro-specifing? Do we need fill issue https://gitlab.archlinux.org/archlinux/packaging/packages/prusa-slicer/-/issues/?sort=created_date&state=opened&first_page_size=20?

edwloef, 2024-12-28:
I would wait for someone who is able to successfully slice the model I referred to (link updated because it was wrong, I had to guess earlier because thingiverse was down) to report to the arch peoples.

ilyxa, 2024-12-28:
Windows 10, fresh install 2.9.0, sliced normal.

edwloef, 2024-12-29:
I meant on another (non-Arch) Linux system. To me there's no indication this is an Arch packaging issue as of now :shrug:

AlexHuebi, 2024-12-29:
I've stumbled across this issue because i wanted to slice another model w/ 0.1mm layer height and my slicer also crashed. 
Although on my project, the error seems to be different
`/usr/include/c++/14.2.1/optional:475: constexpr _Tp& std::_Optional_base_impl<_Tp, _Dp>::_M_get() [with _Tp = Slic3r::Geometry::ArcWelder::Segment; _Dp = std::_Optional_base<Slic3r::Geometry::ArcWelder::Segment, false, false>]: Assertion 'this->_M_is_engaged()' failed.`
`
Aborted (core dumped)
`

I am running EndeavourOS/Arch aswell. 

Because I stumbled over this issue, I also tried to slice it in Linux Mint 22 Liveboot using the Flatpak Slicer version 2.9.0. 
And well.. both my project and @edwloef 's sliced without any issues...

ilyxa, 2024-12-30:
confirmed. flatpack version running well without issue.

archiebug, 2024-12-30:
[code]
/usr/include/c++/14.2.1/bits/stl_vector.h:1213: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::front() [with _Tp = unsigned char; _Alloc = std::allocator<unsigned char>; reference = unsigned char&]: Assertion '!this->empty()' failed.
Aborted (core dumped)
[/code]
I have this issue as well.  Using Arch with prusaslicer from the repo [link](https://archlinux.org/packages/extra/x86_64/prusa-slicer/)

SachCZ, 2025-01-02:
Hello, this is an issue with the way **PrusaSlicer has been compiled for arch by the distribution maintainers**. Even though there is an existing slicer issue causing this to manifest, it **does not happen in the official flatpak version**. We cannot ensure that such issues will not appear in the future until the arch package maintainers do something about it.

PrusaSlicer does not have glibc++ asserts turned on while compiling it for flatpak. Please ask the maintainers of the package to remedy the problem.

We are **in no way opposed** to people distributing the software in a native way on the various linux distributions, but we cannot test them all and these are unofficial (yet highly appreciated) distribution channels. That is why we only **support** flatpak as the official way of using the software on linux.

archiebug, 2025-01-02:
Can anyone open an issue in the arch repo?

https://gitlab.archlinux.org/archlinux/packaging/packages/prusa-slicer/-/issues

edwloef, 2025-01-02:
This may not directly crash on the Flatpak version without assertions, but calling `std::vector::front` on an empty vector still causes UB (see https://en.cppreference.com/w/cpp/container/vector/front), and is still inarguably a bug. I don't think disabling assertions is a solution here.

edwloef, 2025-01-02:
> Thank you, @SachCZ for the explanation.
> 
> Can anyone open an issue in the arch repo?
> 
> https://gitlab.archlinux.org/archlinux/packaging/packages/prusa-slicer/-/issues

This would need to be fixed in the package that owns glibc++ (gcc I think), not prusa-slicer.

archiebug, 2025-01-02:
> > Thank you, @SachCZ for the explanation.
> > Can anyone open an issue in the arch repo?
> > https://gitlab.archlinux.org/archlinux/packaging/packages/prusa-slicer/-/issues
> 
> This would need to be fixed in the package that owns glibc++ (gcc I think), not prusa-slicer.

This should be the package.  It is at v14.2.1 and matches the error message.

gcc - https://archlinux.org/packages/core/x86_64/gcc/

SachCZ, 2025-01-02:
> This may not directly crash on the Flatpak version without assertions, but calling `std::vector::front` on an empty vector still causes UB (see https://en.cppreference.com/w/cpp/container/vector/front), and is still inarguably a bug. I don't think disabling assertions is a solution here.

We obviously know that. That is why I said this:
> Even though there is an existing slicer issue causing this to manifest, it does not happen in the official flatpak version.

I am also not sure about this being a glibc++ "issue" because we are talking about an assert in a header file:
```
/usr/include/c++/14.2.1/bits/stl_vector.h
```
that is probably enabled by declaring `GLIBCXX ASSERTIONS` macro during PrusaSlicer compile time.

In any case, sorry to say, but you are on your own :crossed_fingers:

edwloef, 2025-01-02:
> We obviously know that.

Ah, sorry, in that case I misinterpreted your message.

> that is probably enabled by declaring `GLIBCXX ASSERTIONS` macro during PrusaSlicer compile time.

Ah yes, I totally missed that, since in this case it's not a prusa-slicer specific configuration, but enabled for all package builds by default: [makepkg.conf](https://gitlab.archlinux.org/archlinux/packaging/packages/pacman/-/blob/main/makepkg.conf?ref_type=heads#L47)

archiebug, 2025-01-02:
So, is the resolution updating the makepkg or compiling gcc locally, in an updating makepkg file? How likely would it be to get makepkg updated on the repo?

mplzik, 2025-01-03:
@SachCZ I think there's still some space for misunderstanding, so just to be totally clear, my understaning is that:
- Arch Linux users are on their own when it comes to 2.9.0 crashing and are advised to use the official build that works fine until Arch maintainers fix the package.
- The underlying issue that triggers the assert will be worked on by Prusa engineers and will get fixed in an upcoming release. In that case, is there already any public Github issue to tracking the root cause? (also, by any chance, is this somehow related to https://github.com/prusa3d/PrusaSlicer/issues/11483?)

SachCZ, 2025-01-14:
Hello @mplzik your summary is correct. It is not certain the fix will be in the next release. Also #11483 is the same thing manifesting elsewhere.

lukasmatena, 2025-02-18:
Should be fixed in [PrusaSlicer 2.9.1-alpha1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.9.1-alpha1). Closing.

closed 2025-02-18 by lukasmatena