prusa3d/PrusaSlicer #13861: 2.9.0 specific / crash when open Physical printer tab

ilyxa, 2024-12-28:
### Description of the bug

App crashed when open Physical printer tab

```
[ilyxa@wombatneo ~]$ coredumpctl info 31280
           PID: 31280 (slic3r_main)
           UID: 1000 (ilyxa)
           GID: 1000 (ilyxa)
        Signal: 11 (SEGV)
     Timestamp: Sat 2024-12-28 19:59:39 MSK (5min ago)
  Command Line: prusa-slicer
    Executable: /usr/bin/prusa-slicer
 Control Group: /user.slice/user-1000.slice/session-1.scope
          Unit: session-1.scope
         Slice: user-1000.slice
       Session: 1
     Owner UID: 1000 (ilyxa)
       Boot ID: 333ae6a588af4b0c9583e35fb4862676
    Machine ID: 651f49ec3f874297b4c4f3472cdc661b
      Hostname: wombatneo
       Storage: /var/lib/systemd/coredump/core.slic3r_main.1000.333ae6a588af4b0c9583e35fb4862676.31280.1735405179000000.zst (present)
  Size on Disk: 45.6M
       Message: Process 31280 (slic3r_main) of user 1000 dumped core.

                Stack trace of thread 31280:
                #0  0x00007936119dc523 n/a (libwebkit2gtk-4.1.so.0 + 0x15dc523)
                #1  0x0000793611a1f087 n/a (libwebkit2gtk-4.1.so.0 + 0x161f087)
                #2  0x000079361146235d n/a (libwebkit2gtk-4.1.so.0 + 0x106235d)
                #3  0x000079361182f859 n/a (libwebkit2gtk-4.1.so.0 + 0x142f859)
                #4  0x0000793611931190 n/a (libwebkit2gtk-4.1.so.0 + 0x1531190)
                #5  0x0000793611823135 n/a (libwebkit2gtk-4.1.so.0 + 0x1423135)
                #6  0x0000793611827dad n/a (libwebkit2gtk-4.1.so.0 + 0x1427dad)
                #7  0x000079361015635b n/a (libjavascriptcoregtk-4.1.so.0 + 0x1f5635b)
                #8  0x0000793610156440 n/a (libjavascriptcoregtk-4.1.so.0 + 0x1f56440)
                #9  0x00007936154f9559 n/a (libglib-2.0.so.0 + 0x5d559)
                #10 0x000079361555c257 n/a (libglib-2.0.so.0 + 0xc0257)
                #11 0x00007936154fa287 g_main_loop_run (libglib-2.0.so.0 + 0x5e287)
                #12 0x0000793615be4ebf gtk_main (libgtk-3.so.0 + 0x1e4ebf)
                #13 0x00005e3a152ed546 n/a (n/a + 0x0)
                #14 0x00005e3a153ce8af n/a (n/a + 0x0)
                #15 0x00005e3a151d55ca n/a (n/a + 0x0)
                #16 0x00005e3a146d1dd9 n/a (n/a + 0x0)
                #17 0x00005e3a13efc6f6 n/a (n/a + 0x0)
                #18 0x00005e3a13e37540 n/a (n/a + 0x0)
                #19 0x00007936151e1e08 n/a (libc.so.6 + 0x25e08)
                #20 0x00007936151e1ecc __libc_start_main (libc.so.6 + 0x25ecc)
                #21 0x00005e3a13ec3535 n/a (n/a + 0x0)
                ELF object binary architecture: AMD x86-64
[ilyxa@wombatneo ~]$ coredumpctl info 31337
           PID: 31337 (WebKitWebProces)
           UID: 1000 (ilyxa)
           GID: 1000 (ilyxa)
        Signal: 11 (SEGV)
     Timestamp: Sat 2024-12-28 19:59:46 MSK (5min ago)
  Command Line: /usr/lib/webkit2gtk-4.1/WebKitWebProcess 12 28 31
    Executable: /usr/lib/webkit2gtk-4.1/WebKitWebProcess
 Control Group: /user.slice/user-1000.slice/session-1.scope
          Unit: session-1.scope
         Slice: user-1000.slice
       Session: 1
     Owner UID: 1000 (ilyxa)
       Boot ID: 333ae6a588af4b0c9583e35fb4862676
    Machine ID: 651f49ec3f874297b4c4f3472cdc661b
      Hostname: wombatneo
       Storage: /var/lib/systemd/coredump/core.WebKitWebProces.1000.333ae6a588af4b0c9583e35fb4862676.31337.1735405186000000.zst (present)
  Size on Disk: 34.2M
       Message: Process 31337 (WebKitWebProces) of user 1000 dumped core.

                Stack trace of thread 31337:
                #0  0x000074a37ea4d338 n/a (libnvidia-eglcore.so.565.77 + 0xa4d338)
                #1  0x000074a37e7417e1 n/a (libnvidia-eglcore.so.565.77 + 0x7417e1)
                #2  0x000074a37e778116 n/a (libnvidia-eglcore.so.565.77 + 0x778116)
                #3  0x000074a37e77942f n/a (libnvidia-eglcore.so.565.77 + 0x77942f)
                #4  0x000074a37e746ee8 n/a (libnvidia-eglcore.so.565.77 + 0x746ee8)
                #5  0x000074a37e733b61 n/a (libnvidia-eglcore.so.565.77 + 0x733b61)
                #6  0x000074a37ea34ba8 n/a (libnvidia-eglcore.so.565.77 + 0xa34ba8)
                #7  0x000074a37ea1275d n/a (libnvidia-eglcore.so.565.77 + 0xa1275d)
                #8  0x000074a3a4242aaa n/a (libEGL_nvidia.so.0 + 0x42aaa)
                #9  0x000074a3a4242af9 n/a (libEGL_nvidia.so.0 + 0x42af9)
                #10 0x000074a3a42353ca n/a (libEGL_nvidia.so.0 + 0x353ca)
                #11 0x000074a3a4243e2c n/a (libEGL_nvidia.so.0 + 0x43e2c)
                #12 0x000074a3a42492c6 n/a (libEGL_nvidia.so.0 + 0x492c6)
                #13 0x000074a40774c5c6 n/a (libwebkit2gtk-4.1.so.0 + 0x374c5c6)
                #14 0x000074a403e4e891 n/a (libc.so.6 + 0x3f891)
                #15 0x000074a403e4e95e exit (libc.so.6 + 0x3f95e)
                #16 0x000074a403e34e0f n/a (libc.so.6 + 0x25e0f)
                #17 0x000074a403e34ecc __libc_start_main (libc.so.6 + 0x25ecc)
                #18 0x00006484fb5b7625 n/a (n/a + 0x0)
                ELF object binary architecture: AMD x86-64
[ilyxa@wombatneo ~]$  

```

dmesg -T
```
[Sat Dec 28 19:59:39 2024] slic3r_main[31280]: segfault at 0 ip 00007936119dc523 sp 00007ffdf45c9220 error 4 in libwebkit2gtk-4.1.so.0.16.7[15db523,793611330000+3824000] likely on CPU 8 (core 2, socket 0)
[Sat Dec 28 19:59:39 2024] Code: cc cc cc cc cc cc cc 55 48 89 e5 31 c0 5d c3 cc cc cc cc cc cc cc cc 55 48 89 e5 48 8b 47 18 48 8b 40 28 48 8b b8 68 01 00 00 <48> 8b 07 48 8b 40 10 5d ff e0 cc cc cc 55 48 89 e5 48 83 ec 10 64
[Sat Dec 28 19:59:46 2024] WebKitWebProces[31337]: segfault at 1238 ip 000074a37ea4d338 sp 00007ffdb156a9c0 error 4 in libnvidia-eglcore.so.565.77[a4d338,74a37e239000+c4b000] likely on CPU 4 (core 4, socket 0)
[Sat Dec 28 19:59:46 2024] Code: c1 e0 04 48 8b bc 06 e0 0a 00 00 48 85 ff 74 50 4c 8b a9 70 ab 04 00 83 af 90 00 00 00 01 83 af 8c 00 00 00 01 74 5b 48 89 dd <49> 8b 85 38 12 00 00 4c 89 ef 48 c1 e5 04 48 01 dd 48 8b b0 58 80
```
ArchLinux, latest (nvidia in use). Distro-specific?

### Project file & How to reproduce

open Physical printer tab

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

prusa-slicer 2.9.0-1

### Operating system

archlinux

### Printer model

klipper/moonraker

z0r0, 2024-12-29:
Can confirm that I am also on archlinux with nvidia in use.

ilyxa, 2024-12-30:
archlinux, flatpak 1:1.15.91-1 =  still crashing.

bumbalu, 2025-01-01:
Same issue on linux mint 21.2 Cinnamon. After opening physical printer tab PrusaSlicer freezes and several seconds later crashes.

Jan-Soustruznik, 2025-01-03:
Hello, thank you for your report, It seams that is similar issue of #13835

adejong1952, 2025-01-07:
Same issue here. Manjaro 24.2.1 and Mint Mate 22. Still prints fine as long as you don't click on the Physical Printer tab, then freezes and crashes soon after.

CoffeeSquirel, 2025-02-10:
Can confirm same issue on Fedora 41

NoTreblee, 2025-02-17:
same issue on Arch with Amd gpu

bruno-dogancic, 2025-03-02:
I can also confirm it crashes on Linux Mint 22, Lenovo IdeaPad Gaming 3 (15ACH6) Laptop.

I've tried OrcaSlicer and there I can open MainSail without any problems.

vogon, 2025-06-06:
2.9.2 crashes here for me when I change to the physical printer tab on Fedora 42 with an RTX 2070 and nvidia driver 570.153.02.

```
           PID: 440460 (WebKitWebProces)
           UID: 1000 (vogon)
           GID: 1000 (vogon)
        Signal: 11 (SEGV)
     Timestamp: Thu 2025-06-05 19:55:28 PDT (3min 29s ago)
  Command Line: /usr/libexec/webkit2gtk-4.1/WebKitWebProcess 34 64 66
    Executable: /usr/libexec/webkit2gtk-4.1/WebKitWebProcess
 Control Group: /user.slice/user-1000.slice/user@1000.service/app.slice/app-flatpak-com.prusa3d.Pr>
          Unit: user@1000.service
     User Unit: app-flatpak-com.prusa3d.PrusaSlicer-3116536459.scope
         Slice: user-1000.slice
     Owner UID: 1000 (vogon)
       Boot ID: e8873408532a454c983a0cb15df89e76
    Machine ID: cd0bf51a0db347a6810c8c9ee81462b4
      Hostname: snorvax
       Storage: /var/lib/systemd/coredump/core.WebKitWebProces.1000.e8873408532a454c983a0cb15df89e>
  Size on Disk: 45.2M
       Message: Process 440460 (WebKitWebProces) of user 1000 dumped core.

                Module /usr/lib64/GL/default/lib/libdrm_intel.so.1.124.0 without build-id.
                Module /usr/lib64/GL/default/lib/libdrm_intel.so.1.124.0
                Module /usr/lib64/GL/default/lib/libdrm_amdgpu.so.1.124.0 without build-id.
                Module /usr/lib64/GL/default/lib/libdrm_amdgpu.so.1.124.0
                Module libtinfo.so.6 from rpm ncurses-6.5-5.20250125.fc42.x86_64
                Module libpciaccess.so.0 from rpm libpciaccess-0.16-15.fc42.x86_64
                Module libedit.so.0 from rpm libedit-3.1-55.20250104cvs.fc42.x86_64
                Module libxshmfence.so.1 from rpm libxshmfence-1.3.2-6.fc42.x86_64
                Module libxcb-sync.so.1 from rpm libxcb-1.17.0-5.fc42.x86_64
                Module libSPIRV-Tools.so from rpm spirv-tools-2025.2-2.fc42.x86_64
                Module libxcb-xfixes.so.0 from rpm libxcb-1.17.0-5.fc42.x86_64
                Module libxcb-randr.so.0 from rpm libxcb-1.17.0-5.fc42.x86_64
                Module libxcb-present.so.0 from rpm libxcb-1.17.0-5.fc42.x86_64
                Module libxcb-dri3.so.0 from rpm libxcb-1.17.0-5.fc42.x86_64
                Module libgvfscommon.so from rpm gvfs-1.57.2-1.fc42.x86_64
                Module libgvfsdbus.so from rpm gvfs-1.57.2-1.fc42.x86_64
                Module libcrypt.so.2 from rpm libxcrypt-4.4.38-7.fc42.x86_64
                Module libsasl2.so.3 from rpm cyrus-sasl-2.1.28-30.fc42.x86_64
                Module libevent-2.1.so.7 from rpm libevent-2.1.12-15.fc42.x86_64
                Module libssl.so.3 from rpm openssl-3.2.4-3.fc42.x86_64
                Module libssh.so.4 from rpm libssh-0.11.1-4.fc42.x86_64
                Module libduktape.so.207 from rpm duktape-2.7.0-9.fc42.x86_64
                Module libcurl.so.4 from rpm curl-8.11.1-4.fc42.x86_64
                Module libpxbackend-1.0.so from rpm libproxy-0.5.8-2.fc42.x86_64
                Module libproxy.so.1 from rpm libproxy-0.5.8-2.fc42.x86_64
                Module libmpg123.so.0 from rpm mpg123-1.32.10-1.fc42.x86_64
                Module libogg.so.0 from rpm libogg-1.3.5-11.fc42.x86_64
                Module libopus.so.0 from rpm opus-1.5.2-2.fc42.x86_64
                Module libvorbisenc.so.2 from rpm libvorbis-1.3.7-12.fc42.x86_64
                Module libvorbis.so.0 from rpm libvorbis-1.3.7-12.fc42.x86_64
                Module libFLAC.so.12 from rpm flac-1.4.3-6.fc42.x86_64
                Module libgsm.so.1 from rpm gsm-1.0.22-8.fc42.x86_64
                Module libsndfile.so.1 from rpm libsndfile-1.2.2-8.fc42.x86_64
                Module libgmp.so.10 from rpm gmp-6.3.0-4.fc42.x86_64
                Module libnettle.so.8 from rpm nettle-3.10.1-1.fc42.x86_64
                Module libhogweed.so.6 from rpm nettle-3.10.1-1.fc42.x86_64
                Module libp11-kit.so.0 from rpm p11-kit-0.25.5-5.fc42.x86_64
                Module libblkid.so.1 from rpm util-linux-2.40.4-7.fc42.x86_64
                Module libcrypto.so.3 from rpm openssl-3.2.4-3.fc42.x86_64
                Module libkeyutils.so.1 from rpm keyutils-1.6.3-5.fc42.x86_64
                Module libkrb5support.so.0 from rpm krb5-1.21.3-5.fc42.x86_64
                Module libcom_err.so.2 from rpm e2fsprogs-1.47.2-3.fc42.x86_64
                Module libk5crypto.so.3 from rpm krb5-1.21.3-5.fc42.x86_64
                Module libkrb5.so.3 from rpm krb5-1.21.3-5.fc42.x86_64
                Module libidn2.so.0 from rpm libidn2-2.3.8-1.fc42.x86_64
                Module libunistring.so.5 from rpm libunistring-1.1-9.fc42.x86_64
                Module libvmaf.so.3 from rpm vmaf-3.0.0-3.fc42.x86_64
                Module libcpuinfo.so.24.09.26 from rpm cpuinfo-24.09.26-2.git1e83a2f.fc42.1.x86_64
[... 137 lines cut ...]
                Module libflite.so.1 from rpm flite-2.2-11.fc42.x86_64
                Module libwoff2dec.so.1.0.2 from rpm woff2-1.0.2-22.fc42.x86_64
                Module liblcms2.so.2 from rpm lcms2-2.16-5.fc42.x86_64
                Module libxslt.so.1 from rpm libxslt-1.1.43-1.fc42.x86_64
                Module libxml2.so.2 from rpm libxml2-2.12.10-1.fc42.x86_64
                Module libcairo.so.2 from rpm cairo-1.18.2-3.fc42.x86_64
                Module libatk-1.0.so.0 from rpm at-spi2-core-2.56.2-1.fc42.x86_64
                Module libpango-1.0.so.0 from rpm pango-1.56.3-1.fc42.x86_64
                Module libz.so.1 from rpm zlib-ng-2.2.4-3.fc42.x86_64
[... cut ...]
                Module libwebpmux.so.3 from rpm libwebp-1.5.0-2.fc42.x86_64
                Module libfontconfig.so.1 from rpm fontconfig-2.16.0-2.fc42.x86_64
                Module libfreetype.so.6 from rpm freetype-2.13.3-2.fc42.x86_64
                Module libepoxy.so.0 from rpm libepoxy-1.5.10-9.fc42.x86_64
                Stack trace of thread 8266:
                #0  0x00007fc85d956639 n/a (/usr/lib64/GL/nvidia-570-153-02/lib/libnvidia-eglcore.so.570.153.02 + 0x756639)
                ELF object binary architecture: AMD x86-64
```

Rattkener, 2025-06-06:
Fedora 42, silverblue kinoite 
Prusa Slicer 2.9.2
Trying to load Mainsail + klipper with klipper fw set in physical printer settings. 
Was fine when my printer was off and it showed the error page but when I turned it on and clicked "Reload Firmware" it caused the crashes.

elcalzado, 2025-08-12:
Also on Arch Linux, NVIDIA GPU and I crash whenever I open the physical printer tab. Like Rattkener said, it's fine if printer is off but once it's on, the slicer starts loading the page, freezes, and crashes.

prusa-slicer 2.9.2-11

closed 2026-09-29 by prusaslicer-bot[bot]