prusa3d/PrusaSlicer #13862: 2.9.0. Linux flatpack - No way to select second 'disk'

mw46d, 2024-12-28:
### Description of the bug

I have an Ubuntu 22.04 laptop with two SSDs. Most of my 3d-models are on the second disk but for whatever reason, the version 2.9.0. does not want to 'see' that disk:-( Even when I try to cd into the directory with all the *.stl files, and try to start with one on the command line, the slicer says 'File not found' :-(
The 2.8.1 AppImage did not have that problem. All the permissions work for the last couple of years ... 
I don't know, if that is a feature of the flatpack setup or what's going on ?!

### Project file & How to reproduce

File -> Open -> Other Locations -> Computer  and my /disk1 is not even listed:-(

[image]

```
marcow@feather7:~$ mount | fgrep ' / '
/dev/nvme1n1p7 on / type ext4 (rw,noatime,nodiratime,discard,errors=remount-ro)
marcow@feather7:~$ mount | fgrep ' /disk1 '
/dev/nvme0n1p1 on /disk1 type ext4 (rw,noatime,nodiratime,discard,errors=remount-ro,stripe=8191,data=ordered)
```

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.9.0 Linux flatpack

### Operating system

Ubuntu 22.04 - fully patched

### Printer model

MK4SMMU3, MK3S, Creality Ender3V2

mw46d, 2024-12-29:
```
marcow@feather7:/disk1/marcow/3d_printing$ ps
    PID TTY          TIME CMD
   6190 pts/1    00:00:00 bash
  29001 pts/1    00:00:00 ps
marcow@feather7:/disk1/marcow/3d_printing$ ls -ltr Pi5inchDisplayStand-PartPad.stl
-rw-r--r-- 1 marcow uucp 3284 Oct 25 15:37 Pi5inchDisplayStand-PartPad.stl
marcow@feather7:/disk1/marcow/3d_printing$ flatpak run com.prusa3d.PrusaSlicer ./Pi5inchDisplayStand-PartPad.stl
--------------------------------------------------------------------------
Message: 16:19:16: Starting PrusaSlicer flatpak with entrypoint script
--------------------------------------------------------------------------
marcow@feather7:/disk1/marcow/3d_printing$ No such file: ./Pi5inchDisplayStand-PartPad.stl

marcow@feather7:/disk1/marcow/3d_printing$ ps
    PID TTY          TIME CMD
   6190 pts/1    00:00:00 bash
  29033 pts/1    00:00:00 ps
marcow@feather7:/disk1/marcow/3d_printing$
```

neophyl, 2024-12-29:
Please search for existing issues before opening new ones.  For example your issue sounds like its the same as either this https://github.com/prusa3d/PrusaSlicer/issues/13844 or https://github.com/prusa3d/PrusaSlicer/issues/13820

Both of which appear to be down to the way Flatpack works.  You might want to try https://github.com/prusa3d/PrusaSlicer/issues/13820#issuecomment-2563593643  to see if it can resolve your problem.

mw46d, 2024-12-29:
Thanks. 'Yet Another Permissions System' :-(

Samueru-sama, 2024-12-29:
@mw46d You can also use this [AppImage](https://github.com/probonopd/PrusaSlicer/releases/tag/continuous) from [this relevant issue](https://github.com/prusa3d/PrusaSlicer/issues/13653)

locar82, 2024-12-30:
I'm also having problems with the Flatpak and running a post-processing script using Python. I have given all the necessary permissions, but it refuses to run my script. So, I'm just using the AppImage from the link above, and it runs fine now. Thanks to @probonopd for making this possible!

Jan-Soustruznik, 2025-01-02:
closed as duplicate of https://github.com/prusa3d/PrusaSlicer/issues/13820

closed 2025-01-02 by Jan-Soustruznik