prusa3d/PrusaSlicer #13863: Slicing an object very small pieces of the preview and Gcode are shown on different layer heights

juancanous, 2024-12-28:
### Description of the bug

Note: I think this issue appeared after **upgrading to Prusa 2.9.0**

I have a design that has failed printing a couple of times...this is not a printer issue. After reviewing the preview of the design, I can see there is a Z height change right at the beginning of the layer, meaning, while printing an internal wall at a height of 100mm for example, the external wall is printed at 101mm...this random movement is causing of causing of course the printer to claw into the printing after extruding in the air some filament.
[image]
[image]

### Project file & How to reproduce

Attached you will see the stl file.
Honestly, this is just a simple thing craeted using Onshape, Prusa Slicer 2.9.0
Layers 255 and 509 at 0.2mm.
[Base.zip](https://github.com/user-attachments/files/18267865/Base.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.9.0

### Operating system

WIndows 10

### Printer model

Elegoo Neptune 3

neophyl, 2024-12-29:
Stl file is not much use. What is needed to debug issues are all your settings.  
That's why when opening a new issue you are asked to attach a Prusa Slicer project file that exhibits the issue.  A project file contains not only the part but also all the settings you are using, as well as any custom gcode etc.

I can load your part into one of my printer profiles and it slices perfectly fine.

juancanous, 2024-12-29:
Thanks @neophyl !

First time posting here, too much ignorance about the platform on my end.

I have attached the project...I honestly have never had this type of issue before, not really sure if it's Prusa or directly Onshape exporting the design incorrectly...however, I already exported it as STL and OBJ and both have the same issue.

Thanks again!
[Base.zip](https://github.com/user-attachments/files/18268309/Base.zip)

neophyl, 2024-12-29:
The good news is that I can see the problem when opening your project.  It is definitely not your model file. Nothing wrong with that.  And I mean nothing, took it into some other software to confirm and zero issues detected :)  Which is a rarity :)  

The issue appears to be your After Layer change custom gcode.  When I comment those lines out it slices perfectly.  As to WHY its causing you issues on this file and at those specific areas, that I don't know.  Probably one for the Dev's to look at.

Looks like its switching to relative positioning, moving up 3mm and then going back to absolute positioning and moving somewhere.  I'm guessing its for some kind of time lapse setup ?

I'd have thought if it had a problem with it then it would happen on every layer which it obviously doesn't so there's something else going on.

Anyway hope that helps you get printing that object while waiting for someone who knows more to take a look at it.

juancanous, 2024-12-30:
Thanks @neophyl !

Well, it printed ok, and I saw the printer actually extruding filament in the air and then moving back onto the object and continue to print. Nothing major, but a bit scary to see that happening midprint hehe.

Thanks, and yes, let's see if someone can say something else or if Prusa releases a new update that solves this issue.

Jan-Soustruznik, 2025-01-03:
hello, @juancanous thank you for your report. I created internal issue SPE-2636. We will try to fix in some of next release. 

Thank you @neophyl for you help with investigation. 

Have a nice day.

juancanous, 2025-01-03:
Th and @Jan-Soustruznik for your time in creating the internal issue.

I gotta tell you, I downloaded and tested the previous working version 2.8.1 and the issue disappeared. I am glad older versions are readily available to be able to keep working.

Take care and thanks again!

SachCZ, 2025-01-06:
Hello @juancanous, I am investigating your issue and it is caused by custom after layer change gcode combined with turned off ramping z-hops.

It is still a bug, but the setup is quite advanced and can be fixed by fixing the custom gcode. Was the custom gcode intended?

juancanous, 2025-01-07:
Hi @SachCZ !

Yes, it's a custom Gcode I have in place to take photos for time lapses. This Gcode has been working perfectly for almost one year now. The current version where it works with no issues is 2.8.1, and this is the very first time I see something like this happening.

Thanks Sach!

closed 2025-02-19 by Jan-Soustruznik