prusa3d/PrusaSlicer #13865: Provide "Custom Settings Pages" gathering commonly-changed settings on one pane

davidmcnabnz, 2024-12-29:
**Is your feature request related to a problem? Please describe.**

Prusa 2.9.0 has 15 separate panels of settings, arranged into more than 67 groups, totalling hundreds of separate settings.

This configurability is great, but it brings the problem where finding specific important settings can be very difficult and time-consuming. Also, when going from job to job, it can be easy to forget a couple of crucial settings, which can lead to print failures.

For instance, when moving from one print job to the next, I need to change a handful of settings such as:
* infill
* z-height
* support enabled/disabled
* support structure
* brim on/off, and brim size
* z-lifts
* bed and nozzle temperatures for first and subsequent layers
* speeds for first and subsequent layers

To be fair, several of these commonly-changed settings are present in the Plater tab in the top right panel. But not all.

**Describe the solution you'd like**
It would be really awesome if Prusa allowed the user to create one or more "custom settings panels", where users could group their own most commonly-changed settings into a single simple panel.

**Describe how it would work**

The "Custom Panels Editor" would let the user:
* Create a new panel 
* Give it a name
* Scroll through a tree-structured list of settings (grouped in the hierarchy of pane, group, name)
* Choose desired settings from this tree-list which should be 'mirrored' onto the custom panel
* Move chosen settings widgets up or down, or delete them, or add new ones
* Save the custom panel

The custom panel ideally should also mirror the drop-downs which appear at the top of the panes:
* Print Settings 
* Filaments
* Printers

Then, when moving from one print job to the next, the user could simply load a custom panel and rapidly change whatever settings are needed, with the huge saving of time that comes from having them all together in one place.

**Describe alternatives you've considered**
I have even considered implementing a separate config editor, but this would be clunky and require restarts of Prusa. I'm instead requesting here that a custom settings panels feature be considered and implemented.

**Additional context**
Implementing a Custom Settings Panels Editor would make Prusa far more ergonomic, and help users to streamline their printing workflows, and greatly reduce failures.

neophyl, 2024-12-29:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/8943

davidmcnabnz, 2025-01-01:
It does overlap a lot with #8483 but I wouldn't call it a duplicate. 

Maybe the ideal solution would be to provide elements of both, where users can manage their own pages of commonly-accessed settings, and access these

kubispe1, 2025-01-03:
Thanks for the detailed description of the feature request. We have noted users' needs for such a UI panel.

kubispe1, 2025-01-03:
@davidmcnabnz, why do you think it is necessary to have more panels with a selection of custom parameters? Can you explain your use cases? For example, Cura has just a single parameters view.

davidmcnabnz, 2025-01-04:
The idea is that different users each have their own personal sets of maybe
5-10 "frequently changed parameters".

It takes users a lot of hunting around Prusa's numerous panels to find all
the settings needing changing, and it's easy to forget one or two really
crucial settings.

If users can each collate their own sets of frequently changed settings
into one single custom panel, it would streamline their workflows and
reduce errors and failures.

David

closed 2026-09-29 by prusaslicer-bot[bot]