prusa3d/PrusaSlicer #13866: Build plates

KimA998, 2024-12-29:
In version 2.9.0 it´s been possible to have up to 9 buildplates and thats nice - BUT why ONLY 9 plates???

I have project on my Bambu X1C with 40 different plates based on colors, withs parts belong to each other and so on.

PLEASE MAKE IT POSSIBLE TO HAVE up TO 99 BUILDPLATES...

TIA

Tupson444, 2024-12-31:
I really hope 9 is just a temporary limit. I already have projects where I would use 20+ plates.

(I know that I can split it to multiple files, but the point of multiple plates is to have the same settings in a single file, and reslice everything when tweaking something. If I split to multiple .3mf files, I would need to change settings manually in each file.)

KimA998, 2024-12-31:
And another thing with multiple build plates. Make to possible to assign nozzle and layer size to each plate.

Tupson444, 2024-12-31:
Different settings per plate is suggested in #13858 (as a separate feature request).

UncleJessy, 2025-02-08:
We also need to SNAP to the current buildplate Orca/Bambu that is CRT+0 or CMD+0 and is incredibly useful

gabr42, 2025-10-16:
Yes, please, let us have more than 9 build plates.

lukasmatena, 2026-09-01:
Solved in [3.0.0-alpha11](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_3.0.0-alpha11). Closing.

closed 2026-09-01 by lukasmatena