prusa3d/PrusaSlicer #13867: Multiple-bed number decals float/collide/obscure other mesh

legoadk, 2024-12-29:
### Description of the bug

When there is more than one print bed in a project (using the multiple beds feature), an extra decal is added to each bed to indicate bed number.

This decal is not positioned directly on the bed; it "floats" above the bed, resulting in two issues:

- In the 3D editor view, the decal appears as though it floats and may intersect with objects:
  [image]

- In the Preview view, the decal not only floats as above, but also culls the preview mesh behind it; even though it should be transparent:

  View from above:
  [image]

  View from below:
  [image]

### Project file & How to reproduce

[rc_crawler_whoops.zip](https://github.com/user-attachments/files/18270415/rc_crawler_whoops.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.9.0

### Operating system

Windows 11 - 23H2 (OS Build 22631.4602)

### Printer model

Original Prusa MINI+ (BondTech Extruder)

neophyl, 2024-12-29:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/13814  see comment https://github.com/prusa3d/PrusaSlicer/issues/13814#issuecomment-2558439680

Jan-Soustruznik, 2025-01-03:
cloesed as duplicate of https://github.com/prusa3d/PrusaSlicer/issues/13814

closed 2025-01-03 by Jan-Soustruznik