prusa3d/PrusaSlicer #13872: These STLs crash 2.9.0 on linux

justin-luoma, 2024-12-30:
### Description of the bug

The attached STLs cause crash when importing

### Project file & How to reproduce

[Archive.zip](https://github.com/user-attachments/files/18276159/Archive.zip)

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.9.0

### Operating system

Linux

### Printer model

Sovol SV04

Jan-Soustruznik, 2025-01-02:
Hello, @justin-luoma , could you be more specific how you import the attached stl files? I've tried all available import methods, but none of them cause a crash on my end. I tested this on Ubuntu 22.04.5 LTS.

Jan-Soustruznik, 2025-01-03:
Hello, @justin-luoma, Do you use the oficial Flatpak release? or you use non official linux package. Check comment https://github.com/prusa3d/PrusaSlicer/issues/13859#issuecomment-2568149393 it may help.

closed 2026-09-29 by prusaslicer-bot[bot]