prusa3d/PrusaSlicer #13874: Crash while loading part from SVG in company environment

dt64, 2024-12-30:
### Description of the bug

PrusaSlicer crashs with "boost::filesystem::status illegal function [system: 1]" message while trying to import svg as part etc from Windows server-stored profile.
If same file is loaded from local drive all works fine.

### Project file & How to reproduce

steps to reproduce:
- use Windows 10 Pro in company environment with ADs etc.
- work on a project
- rightclick your object and use "add part -> SVG" 
- select SVG from a server-stored profile folder, e.g. \\server_name.local\my.documents$\Bob.Buttons\My Documents\My Pictures
- enjoy error message:
'Internal error: boost::filesystem::status illegal function [system: 1]: "\\server_name.local\my.documents$\Bob.Buttons\My Documents\My Pictures\test.svg"'

- same file from same location loads without issues if imported in new empty project
- same file loads ok as added part if loaded from a locally mounted drive

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version 2.9.0 

### Operating system

Windows 10 built 19045 64bit

### Printer model

Ender 3

Jony01, 2025-01-02:
Sorry, I can't reproduce your report.

May be, there is some **Access Issues** to you `company environment`
In case, there is different behavior for same file (once it works and once not),
check that file is not locked by another process.

OR

**File System Error**: There might be an underlying problem with the file system, such as corruption or hardware issues.

Till, I do not have enough information to reproduce I can't fix it.

dt64, 2025-01-02:
hm, I can easily reproduce it here with a number of files.
If loaded "normally" it just works, if loaded using "add part" it fails. 

**Workaround:
Copy file to a locally mounted partition and use from there.**

@Jony01 if you can't reproduce in your environment I'm happy to provide additional info, just would need what's required.
Or we just keep it here (maybe closed to be re-opened) in case more people have same experience.

closed 2025-01-03 by Jony01