prusa3d/PrusaSlicer #13875: Filament is wrong message

ac2112, 2024-12-31:
### Description of the bug

The message randomly appears. The filament in my XL is correct, and in prusa slicer is correct. However when I go to print, I constantly get the message "Filament is wrong. The loaded filament does not match the one specified in the print file. Proceed anyway?" 
It doesn't matter what extruder I use.

### Project file & How to reproduce

 I just sliced, selected Sent to Connect and the warning message was shown.

The problem occurred in both 6.2.0-alpha1+8672 and Alpha 2. I'm using Alpha 1 due to a bug in Alpha2

[Warning message.zip](https://github.com/user-attachments/files/18281405/Warning.message.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

6.2.0-alpha1+8672

### Operating system

Windows 11

### Printer model

XL

jnsanders1983, 2025-01-02:
I have the exact same issue on my XL and it does it every time I send it to my printer through connect.

DaveTexarkana, 2025-01-02:
I just installed Prusa SLicer 2.90 (had - Alpha 1 just prior) and this message just started on every send to Connect for me too (Mk4S, Firmware 6.2.0-Alpha2) - has to be bug in 2.90 actual, I guess in combo with Firmware 6.2.0 to be fair.  I haven't had other issues so far with this current Slicer version so will just keep ignoring the error.

This on top of the fairly old error now on the printer, 'Detected multiple load failures....' etc. after every print finishes with no fix or work around published by Prusa (last 2 firmware versions for sure have this, not sure how far back, don't remember, been months) - feels like they're less interested in supporting the bugs they're creating than selling the Core 1 (or creating other nozzle profiles for the Mk4/S + MMU3 as I found many other lamenting).  Please Prusa, get on it? :-?

jnsanders1983, 2025-01-02:
> I just installed Prusa SLicer 2.90 (had - Alpha 1 just prior) and this message just started on every send to Connect for me too (Mk4S, Firmware 6.2.0-Alpha2) - has to be bug in 2.90 actual, I guess [...]

I refuse to use alpha1 or alpha2. I couldn't complete a print because of load errors in alpha1.

DaveTexarkana, 2025-01-02:
Weird such different behavior by installation on core functionality. I’m
using Win11, not a Mac, you?

jnsanders1983, 2025-01-02:
> Weird such different behavior by installation on core functionality. I’m
> 
> using Win11, not a Mac, you?
> 
> 
> 
> 
> 
> 
> > I just installed Prusa SLicer 2.90 (had - Alpha 1 just prior) and this
> 
> > message just started on every send to Connect for me too (Mk4S, Firmware
> 
> > 6.2.0-Alpha2) - has to be bug in 2.90 actual, I guess in combo with
> 
> > Firmware 6.2.0 to be fair. I haven't had other issues so far with this
> 
> > current Slicer version so will just keep ignoring the error.
> 
> >
> 
> > This on top of the fairly old error now on the printer, 'Detected multiple
> 
> > load failures....' etc. after every print finishes with no fix or work
> 
> > around published by Prusa (last 2 firmware versions for sure have this, not
> 
> > sure how far back, don't remember, been months) - feels like they're less
> 
> > interested in supporting the bugs they're creating than selling the Core 1
> 
> > (or creating other nozzle profiles for the Mk4/S + MMU3 as I found many
> 
> > other lamenting). Please Prusa, get on it? :-?
> 
> >
> 
> > I refuse to use alpha1 or alpha2. I couldn't complete a print because of
> 
> > load errors in alpha1.

Windows 11

henryhbk, 2025-01-18:
I too am having the problem on the XL, and clearly it matches, as you can see below:

[image]
filament matches in slicer

[image]
sending via connect from inside slicer, yields the mismatch error.  Unlike most others here I am running totally production 2.9.0 (macOS) and as noted in the screenshot above 6.1.3+7898 firmware.  

[image]
Luckily if you proceed it all works just fine. Fairly sure this is a bug in Connect rather than slicer?

OneEyeLeft, 2025-02-01:
> [image] Luckily if you proceed it all works just fine. Fairly sure this is a bug in Connect rather than slicer?

This is exactly what I experience with my XL. Connect shows correct filament, printer shows correct filament, slicer selected correct filament. This is intermittent to my best recollection and independent of the tool head or filament type chosen. I think this was occurring pre 2.9 slicer and 6.2 firmware.

DaveTexarkana, 2025-02-01:
Is failing in Slicer for me, before I send to Connect, though I am logged
in to connect with in slicer. I should try it logged out and see what
happens.

TastyPi, 2026-01-18:
Is there a way to sync filament configs from PrusaSlicer to the printer? Editing filament configs on the printer is tedious, and I've never successfully got it to say they match. It would be much more convenient to be able to update the printer with the config in PrusaSlicer. Then, the warning could actually be useful, as it would let me know when I accidentally change the settings.

caste1200, 2026-09-02:
> This issues has been labeled **legacy**. It means that it has been created before the release of _PrusaSlicer 3.0.0 alpha_ version. In an attempt to clean up the GitHub issues tracker any **legacy** [...]

Nice, was it ever taken into account? Is it solved on 3.0?

digbat, 2026-09-19:
Still happening: printer PETG - PrusaSlicer PETG - but warning comes:-(

michael-kobb, 2026-09-20:
> Still happening: printer PETG - PrusaSlicer PETG - but warning comes:-(

Are you using the 3.0 beta, or still the 2.x slicer?