prusa3d/PrusaSlicer #13877: Multiple Beds request

Vission01, 2024-12-31:
I really like the new Multiple Beds feature!  I was wondering if it would be possible to have multiple beds support different printers per bed?

I have a Prusa 3.5 a Troodon V2 and a Sovol 04, sometimes I print different pieces of 1 project on multiple printers.  If I could have 1 project file that had each bed with it's own printer and settings that would be amazing.  I notice now that it seems it is set up to only be 1 printer but with multiple beds and the settings are only associated with that printer.

Just a thought,
Thanks!

kubispe1, 2025-01-03:
Hi, thanks for FR.

lukasmatena, 2026-09-01:
Implemented in [3.0.0-alpha11](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_3.0.0-alpha11). Closing.

closed 2026-09-01 by lukasmatena