prusa3d/PrusaSlicer #13878: Option to Temporarily Close Small Holes in the First Layer

tmenzi, 2025-01-01:
[image]
**Feature Request: Option to Temporarily Close Small Holes in the First Layer**

**Description:**  
When printing objects with small holes (e.g., screw holes) using the PrusaSlicer, the first layer often includes many small circular areas corresponding to these holes. This can lead to challenges such as incomplete extrusion, nozzle skips, or adhesion issues, resulting in a failed or suboptimal print.  

To address this, I propose adding a feature that allows users to selectively close small holes or openings in the first layer only. This would create a continuous first layer that prints more reliably. Once the object is printed, these holes could easily be reopened manually, ensuring the model’s intended functionality is preserved.  

**Proposed Workflow:**  
1. **Automatic Detection:** PrusaSlicer detects small openings on the first layer based on a user-defined size threshold (e.g., diameter in mm).  
2. **Selection:** Users can manually review and select which detected holes/openings should be temporarily closed in the first layer.  
3. **First Layer Adjustment:** For selected holes, the slicer generates a continuous first layer by temporarily closing the holes.  
4. **Subsequent Layers:** The original hole geometry is preserved from the second layer onward.  

**Benefits:**  
- Improved first layer adhesion and reliability, especially for prints with multiple small holes.  
- Reduced likelihood of first-layer print failures, saving time and material.  
- Maintains the functionality of the printed model without requiring significant post-processing.  

**Use Cases:**  
- Printing mechanical parts with screw holes.  
- Models with small decorative openings that can cause printing challenges.  
- Any situation where small features on the first layer compromise print quality or success rates.  

**Additional Considerations:**  
- Option for users to toggle this feature on/off for specific holes or globally.  
- A preview mode to visualize closed holes in the first layer before slicing.  
- Configurable threshold for defining what constitutes a "small hole" (diameter and shape).  

This feature would greatly enhance the usability of PrusaSlicer for users printing models with small openings, aligning with its mission to deliver user-friendly and reliable slicing solutions.  

Thank you for considering this request. I'm happy to provide additional feedback or test early implementations if needed!

kubispe1, 2025-01-03:
Hi, thanks for your Feature request. 
I believe that simply "closing" small holes is akin to sweeping garbage under the carpet; it addresses the symptom rather than the root cause. Instead, I propose an alternative solution that could enhance printing precision and quality.

- One possible approach is for the slicer to automatically slow down the speed for small perimeters. We already have a parameter for this in Print Settings > Speed > Small Perimeters, but unfortunately, it does not currently apply to the first layer. I think this limitation should be discussed with the development team to ensure small perimeters on the first layer can benefit from this setting.

- Another potential improvement could involve detecting small holes or loops and scheduling them to print after the infill. (could be simululate by infill_first right now). This way, there would already be some material for the perimeter to bond with, reducing the likelihood of incomplete extrusion or poor adhesion. However, I’m concerned this might impact precision significantly and could result in the hole being partially filled with material, which would require careful evaluation.

tmenzi, 2026-09-19:
Thanks for the suggestions. I think both options are worth trying, but specifically for the first layer. At the very least, these features should be available for the first layer in the slicer.

However, I still miss the option to simply close small holes on the first layer and print it quickly and continuously, then pierce the holes afterwards.

I still have the problem that small holes often don't adhere well and get pulled off, despite a clean build plate and an optimized Z-offset. I have a MKS3+.

Jan-Soustruznik, 2026-10-21:
A similar feature request already exists. This request has been merge to keep topics organized, so please continue the conversation in discussion #15773.

closed 2026-10-21 by Jan-Soustruznik