prusa3d/PrusaSlicer #13880: It's not cool, you don't support Rapberrypi's anymore, not cool at all.

chriswilkins31media, 2025-01-01:
### Description of the bug

It's not cool, you don't support Rapberrypi's anymore, not cool at all. The maker community uses a lot of Rapsberry Pi's.

### Project file & How to reproduce

n/a

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.9

### Operating system

Rapberrypi

### Printer model

custom

sithlord48, 2025-01-01:
Where has support been dropped ? The only distribution for linux is on [Flathub](https://flathub.org/apps/com.prusa3d.PrusaSlicer) and it has both `aarch64` and `x86_64` versions of the application. 

Your gonna need a 64bit pi os so Pi3+ with i would thing the buster based raspian or what they now are calling their version now a days.

SachCZ, 2025-01-02:
Hello @chriswilkins31media, I agree with @sithlord48. There is aarch64 flatpak available.