prusa3d/PrusaSlicer #13881: PrusaSlicer 2..9 HOW to Remove Printables tab ??

DrJuJu, 2025-01-01:
### Description of the bug

With the introduction of v2.9, a "Printables" tab has appeared adjacent to the physical printer !!  how do I move, or preferably remove the unwanted tab ???

### Project file & How to reproduce

Just open application

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.9

### Operating system

Win10, Mate 24.04

### Printer model

Mk2\3.9s\4s

sithlord48, 2025-01-01:
Duplicate: https://github.com/prusa3d/PrusaSlicer/issues/13635 https://github.com/prusa3d/PrusaSlicer/issues/13846

DrJuJu, 2025-01-01:
Appologies for duplicating existing posts, which on a quick search did not highlight themselves

closed 2025-01-06 by TerPuc

kubispe1, 2025-01-06:
@DrJuJu
Thank you for your feedback. We understand that some users are not fully satisfied with the new "Printables" tab introduced in version 2.9.0, and I would like to take this opportunity to clarify our focus.

While we have received significant positive feedback from users who appreciate the convenience and workflow enhancements this feature offers, we acknowledge that not every change aligns with everyone’s preferences. The community surrounding Printables is extensive, and we are confident that many users will find this feature beneficial over time.

This is the first implementation of the "Printables" tab, and our primary focus has been on delivering core functionality. As we move forward, we plan to refine the feature by addressing user concerns, improving organization, and ensuring a smoother, bug-free experience, rather than making every individual button, icon, or tab fully customizable.

Thank you for your understanding.

m00dawg, 2025-02-11:
@sithlord48 Since @kubispe1 was being uncharacteristically and disappointingly unhelpful given Prusa normally has great support, I'll chime in. (I disagree the bug/feature request should have been so flippantly closed and found kubi's response poor given Prusa is ultimately a customer service company first).

You have a couple options. One is OrcaSlicer which is a fork of BambuSlicer which is a fork of PrusaSlicer which is a fork of Slic3r. :P Forks all the way down. Anyways Orca has a different UI which I don't like as much as PS, which makes it a massive shame @kubispe1 was so flippant about the feature since I too find the tab is very annoying and should have a UI option to hide it. Orca doesn't have a Printables (or Bambu's equivalent I forget what it is called) tab along the main menu so this might be a good option and/or you might prefer the UI more than PS. Orca intentionally removes some of the evil Bambu things but there are still some Bambu cloud things in there, though you can turn these off.

Since folks are also having to build AppImage builds because Prusa claims it is difficult to maintain (though community builds show the contrary), the other option is to see if someone forks PS to remove the feature. I'll be looking into this option myself and if I end up with a decent build can share those. I don't plan on having a UI config option though. I was just gonna disable it completely. I like Printables a lot but don't need/want a web browser in my slicer.

m00dawg, 2025-02-11:
Quick update, if building from source, the tab can be hidden by editing src/slic3r/GUI/MainFrame.cpp and commenting out:

```
    add_printables_webview_tab();
```

Here's the exact change in my branch:

https://github.com/m00dawg/PrusaSlicer/blob/c8d2a4fcdecfac421b1128b35dbd76ed7232b462/src/slic3r/GUI/MainFrame.cpp#L809

CharlesScoville, 2025-02-23:
"[...] we have received significant positive feedback from users who appreciate the convenience and workflow enhancements this feature offers [...]"

Let me just stop you right there. I, for one, will not be gaslit into believing that this feature is vocally praised by a strong majority of prusaslicer users. Now I can't speak for everyone, but I can say with conviction that there are a number of obvious and valid reasons that users may not want their slicer to launch an integrated web browser with the click of a single button.

Going further, the fact that this can be patched by merely commenting out a single line of code strongly implies that a trivial GUI checkbox could be added to make this tab entirely optional; making all users happy. The idea that such an option is apparently being categorically rejected is very telling of why it's in here in the first place. . .

This is not about user experience. This is about capturing web traffic.

TL;DR: "Russell's teapot," you are the one making an extraordinary claim about the value of this feature. Show me irrefutable proof that the overwhelming majority of people love it so much that it justifies making it mandatory, or I am deciding PrusaSlicer leadership are "full of it" so cannot be trusted.

Until then, I'm going to call it how I see it. This feature is a prelude to inserting ads into PrusaSlicer.

DaLiV, 2025-04-14:
not only capturing traffic , 
but that is also VIOLATION of GDPR/EU-DSGVO ... ... in princip can SUE the prusa for stealing/selling  personal data to sites which are not related to functional of application ...

neverw1nter, 2025-07-01:
> [@DrJuJu](https://github.com/DrJuJu) ...

As we move forward, we plan to refine the feature by addressing user concerns, improving organization, and ensuring a smoother, bug-free experience, rather than making every individual button, icon, or tab fully customizable.
> 
> Thank you for your understanding.

So is this really the only official answer we've got since January despite people actually having to build the program from source to get rid of the button that they truly think is out of place? What is so damn difficult in adding a simple setting to remove it for those who hates to see it there? What is wrong with you?

neverw1nter, 2025-07-01:
> [@DrJuJu](https://github.com/DrJuJu) ...
>
> As we move forward, we plan to refine the feature by addressing user concerns, improving organization, and ensuring a smoother, bug-free experience, rather [...]

So is this really the only official answer we've got since January despite people actually having to build the program from source to get rid of the button that they truly think is out of place? What is so damn difficult in adding a simple setting to remove it for those who hate to see it there? What is wrong with you?