prusa3d/PrusaSlicer #13882: Multiple nozzle size support not working in this case

bobbyh41, 2025-01-01:
### Description of the bug

My nozzle 1234 are .4 and 5 is a .25, on the top layers I am changing to the .25 nozzle but it’s not actually adjusting the widths. If you select to print the while object with just extruder 5 it makes the adjustments, but not on just the top 2 layers.

### Project file & How to reproduce

[LoveTestMultiProject.3mf.zip](https://github.com/user-attachments/files/18285922/LoveTestMultiProject.3mf.zip)
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.9

### Operating system

M1 Imac 15.1.1

### Printer model

Prusa XL 5 head

kubispe1, 2025-01-09:
Thank you for your feedback. This workflow is not currently available. I understand that this is a desired feature, but it requires more changes to the colors and tool configurations. We will keep this in mind for upcoming updates and aim to modify the current concept to enable this option.

For now, you will need to use the Cut tool to divide the object into parts and assign the tools according to your setup.

kquinsland, 2025-12-24:
Related: https://github.com/prusa3d/PrusaSlicer/issues/13159

closed 2026-09-29 by prusaslicer-bot[bot]