prusa3d/PrusaSlicer #13883: Prusa Slicer drops the authentication way too often

navara, 2025-01-01:
### Description of the bug

During the print, while having Slicer on the second monitor with the Connect page, suddenly without interacting with the app or site, drops the authentication with "Failed to connect to Prusa Account".

There's no retry attempt, no possibility just to refresh the token, nothing - just full reauthentication with name, password (unlocking and fetching from a password manager), finding and unlocking the phone and then MFA app, and providing OTP. Happened to me several times a day a few days ago, and again this time. All while the web session and Android app are working as expected.

Considering how much is Slicer dependent on the Prusa Account to even print (_I'm aware of offline mode with Slicer sending it over WiFi, I consider it a workaround, not a solution_), this is a significant UX issue.

I would expect Slicer to be able to easily re-authenticate, perhaps (optionally) storing long-lasting token to do so on a trusted PC or something like that.

### Project file & How to reproduce

It is not tied to a specific project. It happens randomly with the Slicer opened on the Prusa Connect screen.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.9.0

### Operating system

Windows 11 24H2

### Printer model

Prusa factory built MK4

kocikdav, 2025-01-06:
No extra parameters are needed. There is parameter --loglevel (f.e. --loglevel=5 is the highest level), but in this case default that logs only errors should be fine.
The console should pop up automatically when starting prusa-slicer-console.exe vie icon or start logging if run from console.

halofx, 2025-01-07:
I am definitely experiencing this bug. I routinely work between 2 different machines, and almost always have multiple instances running when dealing with multiple printers in multiple locations. Would love the option to simply say never ask me again, and write the credentials to disk. The whole process of having accounts tied to machines has really broke my work flow.

navara, 2025-01-07:
It doesn't even need to be different machines and printers—every 3mf/stl file opened opens a new instance and possibly kills the session. So, just reviewing a single project and its various files can trigger the bug.

I haven't encountered it since I strictly kept only one instance opened, but it really hampers workflow when viewing/preparing prints…

rcmaniac25, 2025-01-07:
I need to throw my own hat into this. My workflow tends to be opening a PrusaSlicer instance per-project and keeping it open until the project is complete. This sometimes means having 15-20 instances open. One recent project was a multi-bed, painted flexi-dragon that took some 15-30 min to slice.

To spend time working on it, only to find out that PrusaLink is still set to single tool and I can't change that without the model reslicing, and then PrusaConnect has been logged out... I'm basically forced to chose between "USB stick", "log back in" (and I use a password manager, so it's not some tiny password I can "just do"), or restart PS and wait 15-30 min for it to reslice... I often just grab the USB stick because it's faster.

Fixing this issue would mean PrusaConnect actually gets use instead of being a glorified notification and camera system.

raffaellod, 2025-01-21:
Happening to me too, possibly. I logged in once, that lasted maybe 1 day? Let a few days pass, logged in twice, that lasted just about as long. Gave up on the feature since I don’t need it for my workflow (new user, using PrusaLink from day 1), but I imagine I’m missing out on… something, or the in-program login wouldn’t be a thing.

Running PrusaSlicer 2.8.1 on Gentoo/Linux 6.6.9 on amd64.

mittar, 2025-01-21:
Confirm, I have to login multiple times a day, it is crazy. Internet is rocksolid optics 1Gb.
Build: PrusaSlicer-2.9.0
Operating System: Windows
System Architecture: 64 bit

abatkin, 2025-01-30:
This happens to me frequently (sometimes multiple times per day, sometimes I can go a few days with no issue). It's 100% on the Prusa side as I can guarantee that there are no network connectivity issues within my setup (at least nowhere near the frequency it logs me out) since I use a remote desktop and video conferencing software continuously throughout the day so even a second or two or connectivity loss or dropped packets would be immediately apparent.

I have two suggestions:
1) Ignore authentication errors. If I have multiple instances open and one reports the error, I can often exit and re-open the application and I'm still logged in. This tells me that the error is ignorable
2) Use the system browser for logging in (or at least give that as an option), instead of the embedded view as that will result in significantly less friction (since my browser usually already has a valid authentication flow with Prusa, but more importantly I can use my password manager within the browser, so don't need to Copy & Paste things across applications)

In other words, even if Prusa can't/won't fix this bug (that many people have reported across this issue and others), by using an external web browser for login, the bug becomes much less severe.

robojandro, 2025-02-09:
I too would like not to have to log in 4+ times a day.
This happens even if I only have a single PS instance running - I'm using the AppImage on Ubuntu if that helps.
I thought it might be a linux issue and decided to see if someone had created an issue for this behavior and am sad (I guess) to see that it isn't.
I feel like the proper behavior should be not having to log in more than once in a 24 cycle and ideally in perhaps a 7 day cycle.

robojandro, 2025-02-19:
It does appear that 2.9.1-alpha1 does not suffer from the same "log out" issues at 7:30am.
I built it for my Ubuntu system and printed till 4:30pm and was not logged out once during that time, which would make it 9 hours of session time.
It will be interesting to see if the session is still active tomorrow.
I'll report back then.

Jan-Soustruznik, 2025-02-19:
Hello,  Should be fixed in [PrusaSlicer 2.9.1-alpha1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.9.1-alpha1). Closing.

abatkin, 2025-03-27:
I don't know if it's the same issue or some other regression, but I've been running 2.9.1 for a while now, just now when I opened (with no other copies  running) it says "Failed to connect to Prusa Account." in the corner.