prusa3d/PrusaSlicer #13884: Flatpak - missing localizations/translations/languages

cuchac, 2025-01-01:
### Description of the bug

Flatpak 2.9.0 version does not include all localizations. 
Only EN locale is present. I believe this is problem in Flatpak build script. Localization folder (`/app/share/PrusaSlicer/localization`) contains all locales but they are just symlinks to nonexistent `/app/share/runtime/locale/.../...` per-locale folder.

See log from inside flatpak namespace. Symlinks are invalid, for example `../../runtime/locale/cs/cs` does not exists.
```bash
bash-5.2$ pwd
/app/share/PrusaSlicer/localization
bash-5.2$ ls -l
total 0
lrwxrwxrwx 1 nfsnobody nfsnobody 52 Jan  1 09:47 PrusaSlicer.pot -> ../../runtime/locale/PrusaSlicer.pot/PrusaSlicer.pot
lrwxrwxrwx 1 nfsnobody nfsnobody 26 Jan  1 09:47 be -> ../../runtime/locale/be/be
lrwxrwxrwx 1 nfsnobody nfsnobody 26 Jan  1 09:47 ca -> ../../runtime/locale/ca/ca
lrwxrwxrwx 1 nfsnobody nfsnobody 26 Jan  1 09:47 cs -> ../../runtime/locale/cs/cs
lrwxrwxrwx 1 nfsnobody nfsnobody 26 Jan  1 09:47 de -> ../../runtime/locale/de/de
lrwxrwxrwx 1 nfsnobody nfsnobody 26 Jan  1 09:47 en -> ../../runtime/locale/en/en
lrwxrwxrwx 1 nfsnobody nfsnobody 26 Jan  1 09:47 es -> ../../runtime/locale/es/es
lrwxrwxrwx 1 nfsnobody nfsnobody 26 Jan  1 09:47 fi -> ../../runtime/locale/fi/fi
lrwxrwxrwx 1 nfsnobody nfsnobody 26 Jan  1 09:47 fr -> ../../runtime/locale/fr/fr
lrwxrwxrwx 1 nfsnobody nfsnobody 26 Jan  1 09:47 hu -> ../../runtime/locale/hu/hu
lrwxrwxrwx 1 nfsnobody nfsnobody 26 Jan  1 09:47 it -> ../../runtime/locale/it/it
lrwxrwxrwx 1 nfsnobody nfsnobody 26 Jan  1 09:47 ja -> ../../runtime/locale/ja/ja
lrwxrwxrwx 1 nfsnobody nfsnobody 26 Jan  1 09:47 ko -> ../../runtime/locale/ko/ko
lrwxrwxrwx 1 nfsnobody nfsnobody 29 Jan  1 09:47 ko_KR -> ../../runtime/locale/ko/ko_KR
lrwxrwxrwx 1 nfsnobody nfsnobody 38 Jan  1 09:47 list.txt -> ../../runtime/locale/list.txt/list.txt
lrwxrwxrwx 1 nfsnobody nfsnobody 26 Jan  1 09:47 nl -> ../../runtime/locale/nl/nl
lrwxrwxrwx 1 nfsnobody nfsnobody 26 Jan  1 09:47 pl -> ../../runtime/locale/pl/pl
lrwxrwxrwx 1 nfsnobody nfsnobody 29 Jan  1 09:47 pt_BR -> ../../runtime/locale/pt/pt_BR
lrwxrwxrwx 1 nfsnobody nfsnobody 26 Jan  1 09:47 ru -> ../../runtime/locale/ru/ru
lrwxrwxrwx 1 nfsnobody nfsnobody 26 Jan  1 09:47 sl -> ../../runtime/locale/sl/sl
lrwxrwxrwx 1 nfsnobody nfsnobody 26 Jan  1 09:47 tr -> ../../runtime/locale/tr/tr
lrwxrwxrwx 1 nfsnobody nfsnobody 26 Jan  1 09:47 uk -> ../../runtime/locale/uk/uk
lrwxrwxrwx 1 nfsnobody nfsnobody 33 Jan  1 09:47 wx_locale -> ../../runtime/locale/wx/wx_locale
lrwxrwxrwx 1 nfsnobody nfsnobody 29 Jan  1 09:47 zh_CN -> ../../runtime/locale/zh/zh_CN
lrwxrwxrwx 1 nfsnobody nfsnobody 29 Jan  1 09:47 zh_TW -> ../../runtime/locale/zh/zh_TW
bash-5.2$ ls -l ../../runtime/locale/
total 4
drwxr-xr-x 3 nfsnobody nfsnobody 4096 Jan  1  1970 en
```

### Project file & How to reproduce

Just run latest Flatpak 2.9.0 and try to change language.
[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.9.0 flatpak

### Operating system

linux

### Printer model

all

SachCZ, 2025-01-02:
Hello @cuchac,
The application an localization are distributed separately. This is a feature of flatpak. You can configure your flatpak installation to use all the localizations you want.

Here is the relevant information from rc1 changelog:

## Note on localization on Linux

When installing PrusaSlicer using Flatpak, only English and the system language are installed by default. In case you want to switch PrusaSlicer to another language, it is necessary to configure Flatpak to use these languages first. For example

```
sudo flatpak config --set extra-languages "pl_PL;de_DE"
```

will configure Polish and German for Flatpak. It is necessary to reinstall the application after.
Running the command with additional --user parameter will only configure the languages for current user, in which case sudo is not required.

closed 2025-01-02 by SachCZ

cuchac, 2025-01-02:
Hello @SachCZ,
I consider myself a poweruser but still I was unable to discover this (quite hidden) flatpak language-selecting feature/configuration even after proper googling. Even GPT-4o did not come with helpful answer :-)
Would you accept a PR adding some short explanatory text probably with link to details to "change language" dialog when flatpak runtime is detected? This way users like me would not be surprised to find only single language there.

SachCZ, 2025-01-03:
We will think about it :smile: :+1: