prusa3d/PrusaSlicer #13887: Sliced color boundaries from MMU painting are inconsistent compared to Bambu and Orca slicers

jltx1, 2025-01-02:
### Description of the bug

Every few layers, or on repeated layers, the boundary between to colors shifts suddenly.  Though technically still correct, it changes the angle of crossing between the color joints which sticks out like a sore thumb in the actual print and some features are asymmetric.  I have attached some photos where you can see the hat/face, scarf/shoulder, eyes/face are asymmetric and different from the other layers.  The final print looks much worse than the near flawless print that comes off of the Bambu printer.  See side shot as example.   

This issue was in 2.8.1 and maybe earlier, but has gotten worse in 2.9.0.  For example, with this snowman in 2.8.1 I saw 4 bad layers.  In 2.9.0 I got 11 bad layers.  Different models can be even worse.

[image]
[image]
[image]
[image]

### Project file & How to reproduce

open this snowman project and slice.  Use 0.28 layer height if it doesn't load automatically.
[Articulated+Christmas+Collection+–+Snowman.3mf.zip](https://github.com/user-attachments/files/18287446/Articulated%2BChristmas%2BCollection%2B.%2BSnowman.3mf.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.9.0

### Operating system

Mac 15.2

### Printer model

MK3.9S

Jan-Soustruznik, 2025-01-06:
Hello, @jltx1, thank you for your report. We’re already aware of this issue ([#10238](https://github.com/prusa3d/PrusaSlicer/issues/10238)) and are working on a fix. We try to fix in some of new release. 

Have a nice day.

lukasmatena, 2025-02-18:
Should be fixed in [PrusaSlicer 2.9.1-alpha1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.9.1-alpha1). Closing.

closed 2025-02-18 by lukasmatena