prusa3d/PrusaSlicer #13891: Arc fitting is regressing against Bambu and Orca slicers

jltx1, 2025-01-02:
### Description of the bug

While debugging a print quality issue I found that gcode emitted by Bambu and Orca slicers are significantly smaller for the same project.  The main cause is arc "compression".    I sliced both at exact same settings and Bambu found 16730 arcs while PrusaSlicer 2.9.0 only found 10497 arcs.  

I'm not sure how the firmware handles arcs vs line segments but in this case it coincided with a surface quality (external perimeter) difference between the two prints.   There may be some interplay with pressure advance.  Since Prusa locked away the firmware I've stopped looking at the code.

### Project file & How to reproduce

[Articulated+Christmas+Collection+–+Snowman.3mf.zip](https://github.com/user-attachments/files/18292115/Articulated%2BChristmas%2BCollection%2B.%2BSnowman.3mf.zip)

Prusa version:
[PS_Articulated_Snowman_1.3mf.zip](https://github.com/user-attachments/files/18292237/PS_Articulated_Snowman_1.3mf.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.9.0

### Operating system

Mac 15.2

### Printer model

mk3.9s

rcmaniac25, 2025-01-05:
Not sure where you got "Prusa locked away the firmware". It's been https://github.com/prusa3d/Prusa-Firmware-Buddy for years now

jltx1, 2025-01-08:
Now if I make any edits I void my warranty.  That was never the case before and I used to hack it all the time to great effect.  Now it's not worth the risk, so effectively locked up.  Others can disagree but it is a regression.

jmd73, 2025-01-08:
@jltx1 You wont void your warranty any more. You still need to break the tab to be able to flash custom firmware. 

[Core One release blog]https://blog.prusa3d.com/introducing-prusa-core-one-fully-enclosed-corexy-3d-printer-with-active-temperature-control_105477/

It is under the "Make it your own" part.

"Historically, in order to flash custom firmware to all our printers equipped with the Buddy board you had to break a special seal and that meant losing your warranty. Not anymore: we’re making a huge positive change with the aim to make the printer even more community-friendly and motivate developers to dig into the code and come up with all sorts of tweaks and customizations. From now on, breaking the seal won’t void your warranty, so you are free to flash custom firmware."

rcmaniac25, 2025-01-08:
And on top of @jmd73's response, your comment was "I sliced both at exact same settings and Bambu found 16730 arcs while PrusaSlicer 2.9.0 only found 10497 arcs." so it doesn't have to do with firmware anyway. So probably just the settings defaulted in PS print profile vs. whatever Bambu has for their print profile.

jltx1, 2025-01-08:
> @jltx1 You wont void your warranty any more. You still need to break the tab to be able to flash custom firmware.
> 
> https://blog.prusa3d.com/introducing-prusa-core-one-fully-enclosed-corexy-3d-printer-with-active-temperature-control_105477/
> 
> It is under the "Make it your own" part of the blog.
> 
> "Historically, in order to flash custom firmware to all our printers equipped with the Buddy board you had to break a special seal and that meant losing your warranty. Not anymore: we’re making a huge positive change with the aim to make the printer even more community-friendly and motivate developers to dig into the code and come up with all sorts of tweaks and customizations. From now on, breaking the seal won’t void your warranty, so you are free to flash custom firmware."

oh, wow!  thanks for pointing that out.  That is great news.  I'm glad to see Prusa changing their stance.

jltx1, 2025-01-08:
> And on top of @jmd73's response, your comment was "I sliced both at exact same settings and Bambu found 16730 arcs while PrusaSlicer 2.9.0 only found 10497 arcs." so it doesn't have to do with [...]

I'm not clear how you are trying to help.  This is filed on the slicer, not the firmware.  The slicer identifies arc opportunities.  Most of the settings between the two slicers are identical because Bambu forked off PrusaSlicer.  Yes they may have made improvements or Prusa may have made regression since then.  That is what I am highlighting.  I matched the settings between the two, not default settings, and you can play back the gcode and see arcs being stepped out in one case and not in the other.   There is clearly opportunity here, whether is was lost or to be gained, and should be investigated.

rcmaniac25, 2025-01-08:
> I'm not clear how you are trying to help. This is filed on the slicer, not the firmware. The slicer identifies arc opportunities. Most of the settings between the two slicers are identical because [...]

Apologies, but it was clear from the first part of the issue this had nothing to do with firmware and so the second part of the issue discussing firmware effects didn't seem relevant. I simply pointed out, after the firmware portion was addressed, that it could be due to settings differences in the slicers.

As you did this comparison between settings and still have the issue, then I have no extra comments to add.

jltx1, 2025-01-08:
I see.  No worries.  I did make that confusing.  I'm actually less concerned about any file size issues by using more arcs and really curious to see how that may affect print quality.  Since the kinematics can't do native arcs the firmware transforms them into x and y moves.  It *may* be the case that the transformation is atomic with respect to other considerations like pressure advance.  That is, the "unrolling" of the arc into a set of x/y moves is done once along with one PA evaluation, vs evaluating PA for each of the piece-wise linear steppings from the gcode.  As such it may smooth out the extrusion over that arc.  If so then using arc has another advantage wrt quality.  So not a firmware issue but that's why I brought it up.

closed 2026-09-29 by prusaslicer-bot[bot]