prusa3d/PrusaSlicer #13892: Issues with STEP file import

bdeimen, 2025-01-02:
### Description of the bug

Some STEP files end up creating faceted faces instead of curves when imported. I haven't been able to identify what causes the difference. It happened between two different version exports of the same model.
[image]
[image]
[image]

### Project file & How to reproduce

[STEPBugProjects.zip](https://github.com/user-attachments/files/18292488/STEPBugProjects.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.9.0

### Operating system

Windows

### Printer model

Ender 3 Pro

bdeimen, 2025-01-02:
I have re-imported the step file into Fusion and the curves seem to be intact.
[image]
[image]

kubispe1, 2025-01-03:
The way PrusaSlicer handles STEP files is by tessellating the model during import. Tessellation converts the smooth surfaces and curves from the STEP file into a mesh of polygons (typically triangles), which are then used for slicing. This process can sometimes result in faceted faces instead of smooth curves.

bdeimen, 2025-01-03:
That may be the case, but this is still clearly a bug in that process. The same model with only slight alterations imports without the extreme faceting that this one is causes.

bdeimen, 2025-01-03:
Here's another image to clarify what I'm taking issue with:
[image]

closed 2026-09-29 by prusaslicer-bot[bot]