prusa3d/PrusaSlicer #13898: Wipe tower: extraneous extrusion on last layer

brumwald, 2025-01-03:
### Description of the bug

Context: Prusa XL, when printing from more than one filament (which enables the wipe tower) I get a extraneous extrusion on the last layer, even if the the wipe tower was completed many layers ago, resulting in the last (and unnecessary) extrusion to float in the air.

[image]

Was not an issue in v2.8.1 (appimage), but is in v2.9.0 (flatpak).

### Project file & How to reproduce

Example: Benchy_2PLA_MMU3.3mf from https://www.printables.com/model/3110-3d-benchy-2-colors/files

(but this is reproducable on all files that I've tried)

I flipped it upside down on the chimney because then the object ends with just one of the filaments in use and makes the issue more apparent. 
[MK3SMMU2S_Benchy_1.3mf.zip](https://github.com/user-attachments/files/18303306/MK3SMMU2S_Benchy_1.3mf.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.9.0

### Operating system

Linux/Fedora 41 (flatpak)

### Printer model

Prusa XL

neophyl, 2025-01-03:
This is intended operation apparently.  If you had searched both open and closed issues you would have seen 
https://github.com/prusa3d/PrusaSlicer/issues/11904
https://github.com/prusa3d/PrusaSlicer/issues/11692
https://github.com/prusa3d/PrusaSlicer/issues/11397
https://github.com/prusa3d/PrusaSlicer/issues/1355
https://github.com/prusa3d/PrusaSlicer/issues/6568
https://github.com/prusa3d/PrusaSlicer/issues/4005
etc
I'll refer you to this particular comment https://github.com/prusa3d/PrusaSlicer/issues/4005#issuecomment-609945596

rtyr, 2025-01-03:
It's intended even with XL.

closed 2025-01-03 by rtyr